ros-navigation/navigation2 #5213: Enable Pause/Resume in NavigateThroughPoses

tonynajjar, 2025-05-30:
## Feature request

#### Feature description

https://github.com/ros-navigation/navigation2/pull/4994 introduced the concept of WaypointStatus to keep track of the PENDING vs COMPLETED waypoints and that works well for one-shot call to NavigateThroughPoses. 
Typically these missions can run for longer and it would be desirable to pause and resume. 
This feature request is therefore to implement a pause/resume capability.

#### Implementation considerations

We started some implementation discussions already [here](https://github.com/ros-navigation/navigation2/pull/4994#issuecomment-2916001729)

My custom approach:

```
Here is roughly how I have it implemented:

- The navigator client (some kind of mission manager) is responsible for keeping track of the status of all the waypoints by subscribing to the feedback of the navigator.
- My NavigateThroughPoses goal sends `WaypointStatus[]` instead of `nav_msgs/Goals` so that we can pass the statuses
- The navigator's internal `waypoint_statuses` is initialized with the ones provided from the client and only the `goals` that are PENDING are set on the blackboard 
```

Steve's suggestions:

```
Some other thoughts for a hybrid approach:
- We could have a "ResumeNavigateThroughPoses" and "ResumeNavigateToPose" (with complimentary "Pause" ones) so that we have different semantically meaningful APIs for Start/Cancel/Pause/Resume that each have their contextual information. No need to change the main Action APIs, just have new ones that have that information. I think sharper messages are good to have rather than trying to cram everything into one.  
- The Pause can return the state and the Resume can send it back. 
- Or, the Pause can store that in the navigator itself and the Resume can load it back up in a new tree, so that the application doesn't have to care (but can still return it anyway in case they care or want to persist between reboots. If Resume is given values, it uses those instead of its internally stored one).
- Or, the Pause can simply stop ticking the tree to cold store it and the Resume can continue the ticking loop (TBD: some nodes may care about timing or have progress checking, that would need to be considered. Maybe we just `halt` the tree but keep its blackboard active to resume in a fresh new-tick state? That's essentially what your current implementation and the other ideas above do anyway by resume being a re-goal-submission)
```

redvinaa, 2025-06-04:
Hi, me and my team have been using our own solution to pause/resume for quite some time now, it might serve as inspiration.

We implemented a Pause BT control node that wraps the whole tree. It is basically a state machine, and it implements a pause and a resume service. It starts out running the UNPAUSED branch, and halts it on pause, restarts on resume.
To not start everything from scratch each time, we also have a `SequenceWithBlackboardMemory` node, that stores the index of the last child that succeeded on the blackboard (which doesn't get reset on pause). Likewise, all data that we don't want to be lost on pause (halt), we store on the blackboard.

If you find it useful, we can share the code. Here is an example xml with this setup:

```xml
<root BTCPP_format="4" main_tree_to_execute="NavigateToPoseTask">
    <BehaviorTree ID="NavigateToPoseTask">
        <Sequence>
            <Script code="unpaused_branch_child_idx := 0" />
            <Pause pause_service_name="/navigate_to_pose/pause"
                resume_service_name="/navigate_to_pose/resume">

                <!-- UNPAUSED branch -->
                <SequenceWithBlackboardMemory current_child_idx="{unpaused_branch_child_idx}">
                    <!-- First step we don't want to repeat -->

                    <!-- Second super important step -->

                    <!-- Cleanup -->
                </SequenceWithBlackboardMemory>

                <!-- PAUSED branch -->
                <AlwaysSuccess />

                <!-- ON_PAUSE branch -->
                <TriggerEvent service_name="/led_signal/idle" />

                <!-- ON_RESUME branch -->
                <PlaySound service_name="/play_sound" sound="resume" />
            </Pause>
        </Sequence>
    </BehaviorTree>
</root>

```

redvinaa, 2025-06-05:
@SteveMacenski Unfortunately it's not so easy, because we have our custom BT nodes in a separate package. But if it's decided that we're gonna go that way, I can move it to nav2 and open a (draft) PR

SteveMacenski, 2025-06-09:
Great! I think the only thing I'd want from this in addition to the code is some docs on https://docs.nav2.org/behavior_trees/index.html about how to set this up and the design for pause within a BT XML. Your code snippet above as an example would be a good addition (with some more written context). Then we can link that updated BT docs page to the Migration Guide entry mentioning there's now a BT enabled pause / resume feature. That'll gain some discoverability.

Optionally, if you're OK spending a little more time on it, a tutorial would be another option explaining how to modify a BT to have a pause/resume behavior. You can start with one of the Nav2 defaults (your pick), do the migration, and then show the final tree with the pause/resume behavior

SteveMacenski, 2025-08-01:
Implementation 1 inside of the BT is imminent to be merged

closed 2026-09-02 by SteveMacenski