ros-navigation/navigation2 #5233: Conditional Boolean vs String Argument Declaration

nishalangovender, 2025-06-04:
## Feature request

#### Feature description
Whilst using `nav2_bringup`, I noticed that some arguments are defined as native Python booleans (`True`/`False`), whilst others are string-based (`"true"`/`"false"`) depending on the conditional check in the launch file. 

For example, [here](https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L164-L176) you can see `slam` and `use_localization` are in a `PythonExpression` conditional check whilst `use_sim_time`, `autostart` and `use_respawn` are in the `launch_arguments` (funny enough `use_respawn` is defined as `False`).

Whilst this has no real impact on anyone using it, it would be nice to stay consistent with their uses. I'm not sure if I'm in the minority here?

#### Implementation considerations
Possible solution:
`condition=IfCondition(PythonExpression(["'", slam, "' == 'true' and '", use_localization, "' == 'true'"]))`

This keeps everything as string conditions.

SteveMacenski, 2025-06-04:
I think there's some misunderstandings. Lets look at this condition you linked to

> condition=IfCondition(PythonExpression([slam, ' and ', use_localization])),

If you look at the definition of `slam` and `use_localization`, you'll find that they're both `LaunchConfiguration`s that are to be substituted at run-time

> use_localization = LaunchConfiguration('use_localization')

Thus, these are not python booleans. All of our booleans are launch configurations passed into child launch files as launch arguments with perhaps a few exceptions:
* https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L206 We hardcode the container name, though this itself perhaps we should parameterize.
* https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/tb3_simulation_launch.py#L198 keepout / speed zones in the TB3 launch file are disabled since we don't have maps setup for the TB3 sandbox world 

I don't disagree that there are some places where perhaps there are errors that we could fix - can you point some out? 

Grepping through the code, I see some `true` and other `True`s but they're all in strings so that should be fine. I see 2-3x examples of `True` as just a python variable, but they're all related to `use_sim_time` parameter setting in simulation launch files (example: https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/tb4_loopback_simulation_launch.py#L118) so that seems fine and not inconsistent to me

nishalangovender, 2025-06-07:
### Suggestion

Personally I set all of my `launch_arguments` to `true` or `false`:
```
gui = LaunchConfiguration("gui")
declared_argument = [DeclareLaunchArgument("gui", default_value="true", description="Start RViz2 automatically")]
```
and use it in conditions as:
`condition = IfCondition(gui)`

This still allows for the user to set `gui` to `true`, `True`, `false` or `False`.

If I have AND/OR instances (lets say `gui` is `true` and `use_sim_time` is `false`, I use:
`condition=IfCondition(PythonExpression(["'", gui "' == 'true' and '", use_sim_time, "' == 'false'"]))`
(which is the "messiest" use)

Comparatively, an example of a change would be [here](https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L214):

Change: `condition=IfCondition(PythonExpression([use_speed_zones]))`
To: `condition=IfCondition(use_speed_zones)`

SteveMacenski, 2025-06-07:
Doesn’t that have the opposite issue though if someone sets it as ‘True’ instead of ‘true’ due to the string comparison? 

For the single condition checks like speed zones that makes sense and I agree with that to be able to do both. For the multi-conditional checks that gets pretty messy, it would seem, if we need to check both ‘True’ and ‘true’ in all places.

nishalangovender, 2025-06-09:
I would love to! Fair warning this would be my first contribution, but I am definitely ready to contribute to the community 👍

commit bd391296 (2025-06-26): Add LaunchConfigAsBool (Fixes #5233) (#5301)

* Add LaunchConfigAsBool (Fixes #5233)

* Fix Linting

* Fix ament_mypy and pre-commit

* Added Type Annotations

* mypy ignore

* launch.Substitution

* Update All Bools in nav2_bringup

---------
commit b66ef49e (2025-07-17): fix: Add PushRosNamespace action to navigation launch file and update… (#5300)

* fix: Add PushRosNamespace action to navigation launch file and update comments for parameter namespacing

* fix: Reorder import statements in navigation launch file for consistency

* fix: Correct formatting of comments in navigation launch file for clarity

* fix: Add PushRosNamespace action to multiple launch files for parameter namespacing

* fix: Add PushRosNamespace action to slam_launch.py for parameter namespacing

* Adding logging for matched events and dropped messages into pub/sub of new nav2_ros_common [... cut]
closed 2025-06-26 by SteveMacenski