ros-navigation/navigation2 #5267: Rotation Shim with MPPI behaviour different on setting closed_loop param in rotation shim

YKT412, 2025-06-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - Micro-Star International Co., Ltd. GF63 Thin 11UC
- ROS2 Version:
  - Humble
- Version or commit hash:
  - From source: output of `git -C navigation2 rev-parse HEAD` → <insert hash>
  - From apt: output of `dpkg-query --show ros-humble-nav2-rotation-shim-controller` → <insert version>
- DDS implementation:
  - <e.g. Fast DDS, Cyclone DDS>

---

### Steps to reproduce issue

1. Set up Nav2 with the `rotation_shim_controller` and `nav2_mppi_controller`.
2. In your config file, set `closed_loop: false` for the `rotation_shim_controller`.
3. Compare behavior between:
   - Source build of Nav2
   - APT-installed Nav2 (`sudo apt install ros-humble-nav2-*`)
4. Observe differences in behavior of the rotation shim when navigating.

---

### Expected behavior

With `closed_loop: false`, If false, the rotationShimController will use the last commanded velocity as the next iteration’s current velocity. When acceleration limits are set appropriately and the robot’s controllers are responsive, this can be a good assumption. If true, it will use odometry to estimate the robot’s current speed. In this case it is important that the source is high-rate and low-latency to account for control delay.

---

### Actual behavior

- When built from source: `closed_loop` has the expected effect on behavior.
- When installed via APT: `closed_loop` has no apparent effect; rotation shim behavior remains the same regardless of the setting.

---

### Reproduction instructions

- Use the same navigation launch and parameter YAML file in both setups.
- Observe the rotation behavior before and after MPPI starts driving the robot.
- Compare log output or motion trace.

---

### Additional information

It seems like the APT version of `nav2_rotation_shim_controller` might:
- Be outdated
- Not include the `closed_loop` param logic
- Or have it incorrectly wired or disabled

Please confirm whether this is expected due to version mismatch, or if it's a packaging issue.

SteveMacenski, 2025-06-16:
What is the apt version you are using? That was not filled in the issue template. Then, I can look into it. It’s almost certainly just that the binaries need a new sync. This was merged about a month ago after the last sync: https://github.com/ros-navigation/navigation2/pull/5140

Please also describe the behavior that is changed from the binaries versus source

YKT412, 2025-06-18:
apt list --installed | grep ros-humble-navigation2

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-humble-navigation2/jammy,now 1.1.18-1jammy.20250514.173003 amd64 [installed]

SteveMacenski, 2025-09-17:
Closing due to lack of update

closed 2025-09-17 by SteveMacenski