ros-navigation/navigation2 #5274: Converting MPPIController.optimizer_ from value based to pointer to allow inheritance

vahapt, 2025-06-16:
## Feature request

#### Feature description
Currently, `Optimizer optimizer_` is declared as a **value-based object** in the [MPPIController](https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/include/nav2_mppi_controller/controller.hpp) class. This approach does not allow subclasses of `MPPIController` to override or customize the behavior of `Optimizer`. Declaring `optimizer_` as a pointer (e.g. `std::unique_ptr<Optimizer>`) would allow us to replace it.

#### Implementation considerations
There will be an insignificant slowdown due to added pointer lookup (probably a few nanoseconds). But added polymorphic behavior is worth it.

I can create a PR if there are no objections.

vahapt, 2025-06-16:
> any interest in just contributing this to MPPI directly?
There are 3 points I'm considering:
- There's a similar issue #5214 which probably will solve the problem in a better way.
- Adding more variables to a complicated controller increases the learning curve for newcomers, we should keep things as simple as possible.
- I'd love to battle test and improve the algorithm before committing.

Having said that, I'd be more than happy to commit my changes to MPPI directly if you insist. The following parameters should be declared which allows us to configure curvature of the graph:
- wz_std_decay_to
- wz_std_decay_strength

SteveMacenski, 2025-06-16:
I would love to have the contribution. I think it makes sense to add them -- maybe under and `advanced` namespace parameter (i.e. `<controller name>.advanced.wz_std_decay_to` whereas the defaults are essentially disabled (and state that explicitly in the configuration guide alongside the use description). Reducing the learning curve is important, but missing important capabilities takes priority for a production stack like Nav2. At least this way we can mark them as advanced so folks don't mess with them until they're really ready. MPPI has alot of highly technical parameters already so I don't think this is a bridge too far at all.

I'd be happy to field a PR whenever you feel its ready! With respect to the ticket, its still in the air if the `tau`s are going to be added. Recent discussions have pointed to other changes as having had a larger impact so its unclear if that is really going to be needed. So, I wouldn't make decisions based on that for the time being.

vahapt, 2025-06-16:
Agreed. That solves my main issue and one less package to manage on my side :)
I'll create a separate ticket and create my PR regarding the decay function.

Regarding this ticket, we should also introduce some virtual functions to create an entry point for inherited Optimizer class. e.g. `virtual geometry_msgs::msg::TwistStamped evalControl(....)`

I'm generally leaned towards extensibility, flexibility but I also need to understand the team's conventions first. I'll share a separate PR on this and let you/team decide it.

SteveMacenski, 2025-06-16:
W.r.t. the extensibility: these are themselves plugins and usually places that are intentional for extension we make into plugins (so the controller is a plugin, the objective critic functions are a plugin, etc). I don't think of the optimizer being a point of custom configuration for an application, that _should be_ general purpose for all applications. If there are improvements to be made in Optimizer, it usually be best to be contributed upstream so that it can be benefited from all users since I don't think there are any proprietary / highly application specific mods to be made in the Optimizer (unlike the critics whereas specific application constraints may need to be modeled - or the plugins themselves which can be swapped out for the desired robot, environment, or behavior).

I suppose I'm not opposed to making the Optimizer more customizable, but I'd rather do that in response to an issue rather than proactively since I'd like to encourage contributions of that back into the stack. As you showed, those changes are general purposes and can be contributed back to the stack and aren't 'secret sauce' or highly application dependent specializations like the Critics

vahapt, 2025-06-16:
Point very well taken.

vahapt, 2025-06-18:
@SteveMacenski This is getting deeper :)

NoiseGenerator is copying settings_ only when initialize() and reset() functions are called. As I understand, this is done intentionally for thread safety. However it creates the following bug:

When we do a live change `vx_std, vy_std and wz_std` parameters, it is not reflected until the navigation modules are restarted. Tough, it's applied to scoring section live.
```
void NoiseGenerator::initialize(
...
ndistribution_vx_ = std::normal_distribution(0.0f, settings_.sampling_std.vx);
ndistribution_vy_ = std::normal_distribution(0.0f, settings_.sampling_std.vy);
ndistribution_wz_ = std::normal_distribution(0.0f, settings_.sampling_std.wz);
```

From user perspective (I had witnessed it) this creates inconsistent and unexplained behavior.

Reflecting wz_std on every step requires passing some form of referenced variable and updating it in a thread safe way (e.g. when `generateNextNoises()` is called, re-create ndistribution_wz_)

My humble suggestion is to fix the bug while adding the feature. What do you think?

vahapt, 2025-06-18:
I'm getting a bit lost here, `setNoisedControls` is just collecting what has been generated on previous processing loop? So change to "adaptive std's" will always be applied on the next iteration. 

If we agree on the above, `generateNextNoises` feels correct place to process "adaptive std's" to me.

> would just need to be updated to set that "adaptive std's" back to the parameterized std.

We always need the original value of parameterized std to calculate `adaptive` one. So `SamplingStd.wz` will always be parameterized std. `adaptive std` will be another variable.

I forgot to mention that I've switched to `main` branch to avoid massive merge conflicts. I'll cancel current PR and commit with a new (and proper) PR from `main`.
Bug I'm referring may not exist on `jazzy` branch. On  `main` branch, `ndistribution_vx_`, ... are only initialized and not changed even if reset() is called.

> When a dynamic parameter is changed via RQT, it triggers the post params callback

That's enlightening, filled the gap. Thank you.

SteveMacenski, 2025-06-20:
By the way, I noticed the PR is on Jazzy, not `main`. If you're using Jazzy, we can merge that too, but we need a version in main so that all new distributions can have it and I can backport it to Kilted. 

I don't think the `state` needs to contain the `wz_std_adaptive`, that seems like the `SamplingStd` struct makes most sense to manage that. I don't think modifying the state for this feature is desirable. Then have a method `resetAdaptiveStds()` method that for now just does the wz back to the baseline parameter in `reset()`. For `prepare()` we could abstract the name too for `computeAdvaptiveStds(settings, velocity, etc)`. I think both should belong to the `SamplingStd` struct as a logical separation of concerns and more unit testable. `validateWzStdDecayTo` probably belongs there too. 

More pedantic, but the instances of `s.advanced_constraints.validateWzStdDecayTo(s.sampling_std.wz) == false` can just be `!s.advanced_constraints.validateWzStdDecayTo(s.sampling_std.wz)`

 I think your NoiseGenerator would not work. You rmove the creation of the distributions in the initialize function but we call `generateNoisedControls` in some settings that relies on it being around. We add that.

vahapt, 2025-06-20:
Ignore the current PR please. Everything moved to `main` branch already.

About using `state` it still fits better in my brain. But I also understand your thinking. I'll see what I can do.

Other comments are also noted.