ros-navigation/navigation2 #5287: How to configure Smac Planner or costmap for right-side navigation behavior?

brayanpa, 2025-06-18:
### Tuning / Configuration Goal
I would like the robot to prefer navigating on the right side of the path (as a person would), instead of always going through the center. The goal is to make the navigation behavior more socially acceptable and less obstructive, especially in shared environments like hallways.

### What I've Tried
So far, I’ve been handling this using preferred lane costmaps, where I artificially modify the costmap to favor the right side. This works, but it is very tedious because I have to create a separate costmap for every possible path and load it depending on the path.

I also experimented with modifying the planner source code directly. Specifically, I changed the `getTraversalCost()` function in the Smac planner to reduce the cost of right-side movements using a condition like if `(getMotionPrimitiveIndex() == 2)`. While this does push the planner to favor right turns, it also introduces unstable behaviors and causes zigzag motion in many scenarios.

I am now looking for a cleaner or more robust way to achieve right-side preference. My question is: Is there a recommended way to configure the planner or costmap parameters to prefer right-side navigation?

SteveMacenski, 2025-07-03:
@brayanpa any updates / thoughts?

brayanpa, 2025-07-09:
Thanks for the suggestion. I’ll keep working on it as soon as I can.
And yes, my plan is to open a PR once the plugin is complete.

brayanpa, 2025-09-03:
Sorry for the delay — I’ve been really busy with personal matters as I’m moving to a new country. I would still like to complete the plugin and get it merged, but I will only be able to resume work on it in about a month. In the meantime, if anyone is available to take it over and implement the idea, I’d be happy to support and assist.