ros-navigation/navigation2 #5289: The issue of setting the initial point of navigation

QingXiaoYan, 2025-06-20:
I find that the position of the initial point needs to be manually given for each navigation. May I ask if a fixed initial point for navigation can be set in the code

SteveMacenski, 2025-06-20:
Yes, please check out the Action Client -- this is how all navigation goals are sent when you use rviz as a GUI. The selection of what point you'd like to go is up to you / the policy you've set / the application in mind. But it is very easy to set in code. See `nav2_simple_commander` for python examples or the `rclcpp_action::Client` for how to work with it in C++ (they're very analog).

If you have more questions please ask on Robotics Stack Exchange.

closed 2025-06-20 by SteveMacenski

QingXiaoYan, 2025-06-20:
\navigation2-humble\navigation2-humble\nav2_bringup\params\nav2_params.yam

QingXiaoYan, 2025-06-20:
In this file, add the following code to set the initial coordinates for navigation. After enabling the navigation node, it is found that the map cannot be loaded

[image]