ros-navigation/navigation2 #5298: Enable nav2_ros_common interfaces for Services, Subscription, and Action to be lifecycle enabled.

SteveMacenski, 2025-06-23:
Following in the footsteps of the lifecycle publisher, enable the subscription, action server, action client, service server, and service client to have an activate and deactivate method for lifecycle enablement.

These should be added as managed entities and update the Nav2 Task Servers to call `LifecycleNode::on_activate` and complimentary in deactivation so that these are all automatically activated and deactivated behind the scenes. Update the task servers to remove the manual activation and deactivation. We do need to take some care that the instantiation of the interfaces then are ordered since some subscription callbacks might depend on some publishers activation or existence. 

On top, in the nav2 lifecycle node, we should autoactivate if the node is already activated when an interface is created (see `nav2::Publisher` for example)

SuperGops7, 2025-07-03:
Hey @SteveMacenski I'd like to work on this

SteveMacenski, 2025-07-03:
Awesome! I think it makes sense to start off small and do this bit by bit. How about you come up with a game plan and we have some incremental PRs to do this one at a time (subscribers, actions, services, clients, action clients)! Somewhere in there (perhaps after the first?) we should incorporate the boilerplate I mentioned with `activateInterfaces` so that later changes aren't as involved

kailash197, 2025-09-21:
Hi @SteveMacenski I would like to contribute in this issue. @SuperGops7 any update so far?

ar-py-grim, 2025-10-14:
Is somebody also working on it, would love to join them

SteveMacenski, 2025-10-17:
Main runs on Rolling but can be compiled on Jazzy too. We just need to be sure it works on Rolling foremost.

kailash197, 2025-11-06:
@ar-py-grim Lets connect. my discord username is also kailash197

ar-py-grim, 2025-11-06:
@kailash197 i have sent u friend request arpit_1348

Prathmesh2931, 2025-12-04:
@SteveMacenski  ,
Is it okay to start contributing on the below issue, like  both  of them are not responding .

SteveMacenski, 2025-12-04:
Please 😄

Lotusymt, 2026-01-03:
Hi @SteveMacenski,

I implemented the **subscription** piece in pr #5834.

**Plan (subscriptions):**

* Introduce `nav2::Subscription<T>` as a **lifecycle-enabled managed entity** (similar to `rclcpp_lifecycle::LifecyclePublisher`), so it can be activated/deactivated.
* Keep the actual transport as an underlying `rclcpp::Subscription<T>` internally, but **wrap the user callback** to gate execution on `is_activated()`:

  * If not activated: drop messages + `WARN` once per inactive period.
  * If activated: run the user callback normally.
* For **non-lifecycle `rclcpp::Node` usage**, I added a one-time constructor check and **auto-activate** (so behavior matches legacy “subscribe works immediately” and avoids breaking plain-node use cases / tests).

**Tests:**

* I made a small adjustment to `test_actions` to align with the new subscription behavior/types, and verified all tests pass.

**Next steps after this draft PR:**

* Repeat the same pattern for **services/clients/action servers/action clients** incrementally.
* After those land, follow up with the `LifecycleNode::activateInterfaces()` ordering work (exporters before consumers; deactivate in reverse), and then remove the manual activation boilerplate in task servers.
* Finally, add the “autoactivate if node already active when interface is created” behavior across the remaining interfaces (matching `nav2::Publisher`).

Also looping in @Prathmesh2931 / @ar-py-grim / @kailash197 — I’d love to collaborate. Feel free to comment on the draft PR (or share progress/areas you’re working on) so we can avoid overlap and split the remaining interface types cleanly.

Thanks!

ar-py-grim, 2026-01-05:
@Lotusymt i will be happy to collaborate. let me know how should we communicate

Lotusymt, 2026-01-08:
> Sure [@Lotusymt](https://github.com/Lotusymt), let me know how should we got about dividing task

For now, I think it would be good to first agree on how to implement the subscription part. I’ve made an initial attempt, but I’m not fully confident that this approach is viable yet. If you have time, could you also take a look or give it a try on your side, and share your thoughts on the questions I raised above?

Lotusymt, 2026-01-13:
Thanks @SteveMacenski ! Glad to hear that 🙂

If it makes sense, would it be okay to spin up a small private Slack channel within the Nav2 workspace? We’ll of course summarize any decisions back on GitHub once things are clearer.

I also wanted to briefly clarify my current **design direction** #5834  and ask for guidance before pushing this further:

- My goal is to keep `nav2::interfaces::create_subscription` **usable for any node type**, similar to the existing publisher / client / action patterns.
- To support lifecycle behavior safely, I added a small **friend hook in `nav2::LifecycleNode`** that allows the lifecycle overload of `nav2::interfaces::create_subscription` to call `add_managed_entity()` *only when the node is actually a lifecycle node*.  
  This way, even if a user directly uses the interface factory with a lifecycle node, the subscription can still be registered correctly.
- For non-lifecycle nodes, the `nav2::Subscription` is auto-activated to preserve backward compatibility.

The PR builds successfully, but as expected, some release / CI tests are not passing yet. I haven’t started addressing those since I wanted to first confirm that this overall direction is reasonable before propagating changes across the codebase.

Does this approach seem plausible to you, or would you prefer a stricter separation (e.g. lifecycle-only subscription factories)? Any guidance here would be really helpful before I continue.

Lotusymt, 2026-01-13:
> I think you should be able to create a new channel - I don't think though it needs to be private :-)
> 
> I think the managed entity work can be handled down the line since we don't have this for actions/services/clients yet. But, I did take a look and it seems generally reasonable as a starting point. It looks a bit WIP so I don't want to scrutinize too much at this point. For example I see 3x `create_subscription`s in the factories that I assume would be reduced back to 1 & a few odd `interface` namespace implementations bleeding into the lifecycle_node.hpp file. I assume (?) that there's nothing special about the Subscriptions that we cannot do with LifecyclePublisher to require having too much additional complexity outside of its own file.
> 
> I will say that you have a `should_log_` that could be replaced by just using `RCLCPP_WARN_ONCE` macro. Though I think maybe this should happen more than once 😉

Thanks, Steve! 

I agree there’s likely nothing subscription-specific here, and I can implement `nav2::Subscription` to mirror `LifecyclePublisher` closely. I just realized that it can function properly cause the activation behavior is essentially handled explicitly on the Nav2 side (or via the lifecycle transition plumbing), regardless of whether it got added to the managed-entity list or not.

However, one small edge case I noticed (from the existing “helper/utility classes accept any node type” pattern, e.g. `OdomSmoother`, `FootprintSubscriber`): those helpers call the generic `nav2::interfaces::create_subscription(...)` with arbitrary node types. In that path, the subscription will **not** be registered as a managed entity (even if the passed node is actually a lifecycle node).

If you’re OK with **not guaranteeing managed-entity registration** when users bypass `nav2::LifecycleNode::create_subscription()` and call the generic factory directly, I’ll proceed with the simpler `LifecyclePublisher`-style implementation and keep registration only in the lifecycle node path.