ros-navigation/navigation2 #5299: Update declares and gets to use new declare_or_get_param API

SteveMacenski, 2025-06-23:
The new `nav2_ros_common` package has a `declare_or_get_param` API from @MarcoMatteoBassa that we should use instead of the usual pattern we have for `declare_parameter_if_not_declared` and then `get_parameter`. That way we declare if not declared and return the value all on a single line. 

This should become a member of the `nav2::LifecycleNode` and we can use it from the `node->declare_or_get_param` or as standalone `nav2::declare_or_get_param` when given a `NodeT` or parameters interface.

SteveMacenski, 2025-06-25:
Future steps
- [ ] Make it a member of `nav2::LifecycleNode`
- [ ] Audit remaining parameter usage to see if anywhere else now using the main parameter API could benefit from this

elsayedelsheikh, 2025-06-28:
I'd like to handle this 😃

SteveMacenski, 2025-07-01:
Ok, please! That would be much appreciated!

elsayedelsheikh, 2025-09-26:
> Hey Sayed,
> I was looking to contribute to the migration process for the new API.
> I will open some PRs against the unticked packages and update them once merged.

No problem, Go ahead!
Thanks

mini-1235, 2025-10-01:
@leander-dsouza Just an FYI — I am refactoring the controller server to use this API in #5446

SteveMacenski, 2025-10-17:
Nope, you are welcome to it!

SteveMacenski, 2025-11-14:
@Hobanghann I merged yours for NavFn, feel free to do a few others if you would like! The help is appreciated :-)