ros-navigation/navigation2 #5324: Would a planner recovery help - ComputePathThroughPoses

mrmara, 2025-07-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubunt 22.04
- ROS2 Version:
  - humble
- Version or commit hash:
  - humble_main
- DDS implementation:
  - Cyclone

#### Steps to reproduce issue
Would a planner recovery help BT node is not working for ComputePathThroughPoses action as it does for ComputePathToPose. This is probably due to a typo:

[CODE](https://github.com/ros-navigation/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/plugins/condition/would_a_planner_recovery_help_condition.hpp
)

SteveMacenski, 2025-07-02:
@claude fix this typo

SteveMacenski, 2025-07-02:
Worked on this one! Closing and thanks for the report!

commit e841c7bd (2025-07-02): fix: correct ThroughActionResult type alias in would_a_planner_recovery_help_condition

The ThroughActionResult type alias was incorrectly referencing Action::Result 
instead of ThroughAction::Result, causing the condition to not work properly 
for ComputePathThroughPoses actions.

Fixes #5324
commit 347afe8d (2025-07-02): fix: correct ThroughActionResult type alias in would_a_planner_recovery_help_condition (#5326)

The ThroughActionResult type alias was incorrectly referencing Action::Result 
instead of ThroughAction::Result, causing the condition to not work properly 
for ComputePathThroughPoses actions.

Fixes #5324