ros-navigation/navigation2 #5328: Nondescript issue building from source on Humble branch

scalidas, 2025-07-03:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04 on WSL
- Computer:
  - AMD Ryzen 7 Laptop CPU, 8GB RAM
- ROS2 Version:
  - Humble from binaries
- Version or commit hash:
  - 623636cef3c6f29ac2538aa28f31278a5dd7b54f
  - Most recent commit on humble branch
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
Following the build and install guide given on https://docs.nav2.org/development_guides/build_docs/index.html#build
```
mkdir custom_nav2_ws/src
cd custom_nav2_ws/src
git clone git@github.com:ros-navigation/navigation2.git --branch humble
rosdep install -y   --from-paths ./src   --ignore-src
colcon build --symlink-install
```

Also tried 
`colcon build --allow-overriding nav2_bringup nav2_costmap_2d nav2_map_server nav2_msgs navigation2 --executor sequential` to compile one at a time to avoid running out of memory.

#### Expected behavior
Compile without error

#### Actual behavior
`colcon build --symlink-install`

```
Starting >>> nav2_common
Starting >>> nav_2d_msgs
Finished <<< nav2_common [1.31s]
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid
Finished <<< nav_2d_msgs [8.23s]
Starting >>> dwb_msgs
Finished <<< nav2_voxel_grid [26.9s]
Finished <<< dwb_msgs [20.2s]
Finished <<< nav2_msgs [42.4s]
Starting >>> nav2_util
Starting >>> nav2_simple_commander
Finished <<< nav2_simple_commander [1.72s]
[Processing: nav2_util]
Finished <<< nav2_util [42.1s]
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server
Starting >>> nav_2d_utils
Starting >>> nav2_behavior_tree
Starting >>> nav2_amcl
Starting >>> nav2_velocity_smoother
[Processing: nav2_amcl, nav2_behavior_tree, nav2_lifecycle_manager, nav2_map_server, nav2_velocity_smoother, nav_2d_utils]
--- stderr: nav2_velocity_smoother
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/velocity_smoother_core.dir/build.make:76: CMakeFiles/velocity_smoother_core.dir/src/velocity_smoother.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:174: CMakeFiles/velocity_smoother_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_velocity_smoother [38.4s, exited with code 2]
Aborted  <<< nav2_amcl [38.4s]
Aborted  <<< nav2_lifecycle_manager [54.8s]
Aborted  <<< nav_2d_utils [2min 35s]
Aborted  <<< nav2_behavior_tree [2min 35s]
Aborted  <<< nav2_map_server [2min 35s]

Summary: 7 packages finished [4min 2s]
  1 package failed: nav2_velocity_smoother
  5 packages aborted: nav2_amcl nav2_behavior_tree nav2_lifecycle_manager nav2_map_server nav_2d_utils
  6 packages had stderr output: nav2_amcl nav2_behavior_tree nav2_lifecycle_manager nav2_map_server nav2_velocity_smoother nav_2d_utils
  26 packages not processed
```

`colcon build --allow-overriding nav2_bringup nav2_costmap_2d nav2_map_server nav2_msgs navigation2 --executor sequential`

```
Starting >>> nav2_common
Finished <<< nav2_common [0.80s]
Starting >>> nav_2d_msgs
Finished <<< nav_2d_msgs [0.72s]
Starting >>> dwb_msgs
Finished <<< dwb_msgs [0.77s]
Starting >>> nav2_msgs
Finished <<< nav2_msgs [0.82s]
Starting >>> nav2_voxel_grid
Finished <<< nav2_voxel_grid [0.28s]
Starting >>> nav2_simple_commander
Finished <<< nav2_simple_commander [1.52s]
Starting >>> nav2_util
Finished <<< nav2_util [0.37s]
Starting >>> nav2_amcl
Finished <<< nav2_amcl [36.2s]
Starting >>> nav2_behavior_tree
--- stderr: nav2_behavior_tree
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [test/plugins/condition/CMakeFiles/test_condition_is_stuck.dir/build.make:76: test/plugins/condition/CMakeFiles/test_condition_is_stuck.dir/test_is_stuck.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2069: test/plugins/condition/CMakeFiles/test_condition_is_stuck.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [test/plugins/condition/CMakeFiles/test_condition_is_battery_charging.dir/build.make:76: test/plugins/condition/CMakeFiles/test_condition_is_battery_charging.dir/test_is_battery_charging.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2097: test/plugins/condition/CMakeFiles/test_condition_is_battery_charging.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_behavior_tree [2min 48s, exited with code 2]

Summary: 8 packages finished [3min 30s]
  1 package failed: nav2_behavior_tree
  1 package had stderr output: nav2_behavior_tree
  30 packages not processed
```

#### Reproduction instructions
See above

#### Additional information
I find it hard to believe this is an issue with nav2 itself, but I'm unable to get any sort of hint at what the issue is. The apt installed binaries work fine and I have had no other issue compiling ROS packages. 

I tried running colcon build with `--event-handlers console_direct+ --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON` and while it produced a lot more output, none of it seemed to have any indication on the source of the error. My computer was at a very high RAM usage though - could this be the problem?

SteveMacenski, 2025-07-03:
That’s the pretty classic ‘I ran out of resources’ error. That has nothing to do with ROS, Nav2, etc.

scalidas, 2025-07-04:
You guys are correct - it was running out of memory. I was always running only one job at a time with `--executor sequential`. I had to close all other windows and ran it a few times and managed to get it to build.