ros-navigation/navigation2 #5330: SMAC smoother can generate infeasible paths

tonynajjar, 2025-07-03:
Following up on https://github.com/ros-navigation/navigation2/issues/4271#issuecomment-2879364973

The SMAC smoother could result in an infeasible (colliding) path. The reason is because after modifying the path, it does simple cell cost collision checking (rejects path if cell cost > MAX_NON_OBSTACLE) but does not check for potential collision of orientated footprints.

SteveMacenski, 2025-07-22:
The issue here is that we don't have the feasibility information in the smoother currently. So, there'd have to be a check on the feasibility constraints to then be able to get the orientation information (theta) needed to check the footprint. We cannot check the footprint without X,Y,Theta for a pose since the footprint is orientation-dependent when non-circular.

So, we'd probably want to start off by estimating the orientation with the `updateApproximatePathOrientations` function to get an orientation for introducing the collision checker. Then we'd have the X,Y,Theta for each pose to use a collision checker object with (Nav2 Costmap 2D has one of these you can use, see Hybrid-A* NodeT for an example use). 

Then, perhaps trying to do one step better by either changing the way the smoother works to retain orientation information so we know its 'real' or make sure we maintain minimum turning radius constraints so that the computed orientation is known to be drivable. That way if not just collision free, but also doesn't possibly violate constraints (which is rare and not usually a big deal, but can happen and would impact vehicles like Ackermann that have an actual, strict limit)

SteveMacenski, 2025-08-12:
Yes, that sounds like a good plan! I don't have any notes :-)