ros-navigation/navigation2 #5358: setting recovery_alpha_fast and recovery_alpha_slow throws segmentation fault

sukruthi-chidananda, 2025-07-14:
`amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.5
    alpha3: 0.7
    alpha4: 0.25
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5 #increase to 5.5 for large env
    beam_skip_error_threshold: 0.8
    beam_skip_threshold: 0.25
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 10.0
    laser_max_range: 12.0
    laser_min_range: 0.05
    laser_model_type: "likelihood_field_prob"
    max_beams: 60
    max_particles: 2000
    min_particles: 800
    odom_frame_id: "odom"
    pf_err: 0.1
    pf_z: 0.95
    recovery_alpha_fast: 0.01 #making it 0.1 causing seg fault
    recovery_alpha_slow: 0.001 #making it 0.001 causing seg fault
    resample_interval: 5
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 5.0
    # service_introspection_mode: "disabled"
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.7
    z_max: 0.05
    z_rand: 0.3
    z_short: 0.05
    scan_topic: scan
    # publish_visualizations: true
    set_initial_pose: true
    always_reset_initial_pose: false
    first_map_only: false
    initial_pose:
      x: 0.0
      y: 0.0
      z: 0.0
      yaw: 0.0

amcl_map_client:
  ros__parameters:
    use_sim_time: False
    sigterm_timeout: 10
    sigkill_timeout: 20

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False
    sigterm_timeout: 10
    sigkill_timeout: 20

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: "map"
    robot_base_frame: "base_footprint"
    odom_topic: "odom"
    bt_loop_duration: 10
[... 330 lines cut ...]
waypoint_follower:
  ros__parameters:
    use_sim_time: False
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
    sigterm_timeout: 10
    sigkill_timeout: 20`

Error:
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
Jul 11 01:31:34 ev2-3base docker[4742]: [map_controller-11] [ERROR] [component_container_isolated-1]: process has died [pid 5154, exit code -6, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_omppq52r --params-file /tmp/launch_params_o7_zkwq9 -r /cmd_vel:=/cmd_vel_nav -r /tf:=tf -r /tf_static:=tf_static'].

Attached gdb to the isolated container and found that amcl crashes and map->odom vanishes, AMCL is trying to access memory that doesn't exist.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000ffffa6652aac in __GI_abort () at abort.c:79
#2  0x0000ffffa077546c in ?? () from /lib/libGraphicsMagick-Q16.so.3
#3  <signal handler called>
#4  0x0000ffff80bee390 in nav2_amcl::AmclNode::uniformPoseGenerator(void*) () from /opt/ros/humble/install/lib/libamcl_core.so
#5  0x0000ffff80ad7e38 in pf_update_resample () from /opt/ros/humble/install/lib/libpf_lib.so

SteveMacenski, 2025-07-14:
Are you setting it to legitimate values? Please provide instructions to reproduce

SteveMacenski, 2025-09-18:
Closing due to lack of response

closed 2025-09-18 by SteveMacenski