ros-navigation/navigation2 #5375: MPPI Controller tuning tips to fix bad performance on differential drive robots in close maneuvering goals

vignesh-anand, 2025-07-20:
Required Info:

Operating System & version: Ubuntu 22.04
Computer: NVIDIA Jetson Orin Nano
ROS 2 distribution & install method: Humble (binaries)
Navigation2 version/commit:
• ros-humble-navigation2 1.1.17-1jammy.20241128.061340
• ros-humble-nav2-mppi-controller 1.1.17-1jammy.20241128.041041
DDS implementation: Fast DDS
Nav2 package in use: MPPI controller
Have you asked this on Robotics Stack Exchange? [Yes / No; if yes, paste the link]

Tuning / Configuration Goal:

My tuning goal is for the robot to successfully reach setpoints in close vicinity (< 0.5 m), using whatever behavior (pivot, forward/backward arc, etc.) is necessary to get there reliably.

Here is a video i recorded of the current bad behavior. Eventually the controller fails and times out trigger a backup, before taking over again, failing again and backing up again

Attaching my controller config
controller_server:
  ros__parameters:
    use_sim_time: true
    controller_frequency: 20.0
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.5         # Differential drive: no lateral motion
    min_theta_velocity_threshold: 0.001
    progress_checker_plugins: ["progress_checker"]
    goal_checker_plugins: ["goal_checker"]
    controller_plugins: ["FollowPath"]
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.25
      yaw_goal_tolerance: 0.1
      stateful: True

    # MPPI Controller (rotation shim removed)
    FollowPath:
      plugin: "nav2_mppi_controller::MPPIController"
      
      # MPPI controller parameters
      motion_model: "DiffDrive"
      time_steps: 50                    # 50 steps * 0.05s = 2.5s prediction horizon
      model_dt: 0.05                    # 1/controller_frequency = 1/20 = 0.05s
      batch_size: 1000                  # Number of trajectory samples
      iteration_count: 1
      
      # Velocity limits
      vx_min: -0.2
      vx_max: 0.5
      vy_max: 0.0                       # differential drive - no lateral motion
      wz_max: 0.8
      
      # Acceleration limits
      ax_min: -0.3
      ax_max: 0.3
      ay_min: 0.0                       # differential drive - no lateral acceleration
      ay_max: 0.0
      az_max: 0.5
      
      # Sampling standard deviations
      vx_std: 0.2
      vy_std: 0.0                       # No lateral sampling for diff drive
      wz_std: 0.4
      
      # Path handling
      transform_tolerance: 0.2
      prune_distance: 1.5
      max_robot_pose_search_dist: 10.0
      
      # Algorithm parameters
      temperature: 0.3
      gamma: 0.015
      visualize: true
      regenerate_noises: false
      
      # Goal tolerances
      xy_goal_tolerance: 0.25
      
      # Critics (objective functions)
      critics: [
        "ConstraintCritic", 
        "CostCritic", 
        "GoalCritic", 
        "GoalAngleCritic", 
        "PathAlignCritic", 
        "PathFollowCritic", 
        "PathAngleCritic", 
        "PreferForwardCritic"
      ]
      
      # Constraint critic
      ConstraintCritic:
        enabled: true
        cost_power: 1
        cost_weight: 4.0
      
      # Cost critic
      CostCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.8
        critical_cost: 300.0
        consider_footprint: true
        collision_cost: 1000000.0
        near_goal_distance: 0.5
        trajectory_point_step: 2
      
      # Goal critic
      GoalCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        threshold_to_consider: 1.2
      
      # Goal angle critic
      GoalAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.0
        threshold_to_consider: 0.5
      
      # Path alignment critic
      PathAlignCritic:
        enabled: true
        cost_power: 1
        cost_weight: 14.0
        max_path_occupancy_ratio: 0.05
        trajectory_point_step: 4
        threshold_to_consider: 0.5
        offset_from_furthest: 20
        use_path_orientations: false
      
      # Path following critic
      PathFollowCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        offset_from_furthest: 6
        threshold_to_consider: 1.2
      
      # Path angle critic
      PathAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 2.2
        offset_from_furthest: 4
        threshold_to_consider: 0.5
        max_angle_to_furthest: 1.0
        mode: 0
      
      # Prefer forward critic
      PreferForwardCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
        threshold_to_consider: 0.5

Things I have tried, high controller frequency (30 Hz, 50 Hz) longer prediction windows (all the way to 200 steps).  Batch size 3000. Path follow and path align critics weight higher, reducing distance thresholds. Would appreciate

https://github.com/user-attachments/assets/2747b1cd-0762-4f3d-8a5a-2e12466ee6aa

 advice on tuning ideas?

SteveMacenski, 2025-07-22:
MPPI (and DWB or other controllers) are going to require some tuning to get the particular behavior you're looking for. I give some advise in the README for how I think about tuning MPPI which should hopefully help https://github.com/ros-navigation/navigation2/tree/main/nav2_mppi_controller#notes-to-users

Its difficult from afar for me to help with detailed tuning, it requires alot of visualization, testing things out, and understanding your specific requirements. Generally, we don't provide tuning help in the issue tracker for that reason and limited open-source maintainer cycles. I hope that the documentation there can give you a good head start and I'm happy to answer a few, specific questions if there are specific questions you can prose.

how3v3r, 2025-09-22:
That issue occurred to me either. Differential model robot cannot correct its lateral error around the goal position.
Did you solve this problem by tuning? Or just use another controller to get to the position precisely?