ros-navigation/navigation2 #5380: Best way to pass externally generated NURBS path as global plan in Nav2

Calcifer852, 2025-07-22:
I am setting up a simple VDA5050 vehicle where the fleet manager provides the path as a NURBS curve. What is the recommended way to pass this non-discretized NURBS path from my external node to the Nav2 planner as the global plan? Is there a supported interface for this in Nav2, or does the stack only accept sampled (discretized) paths?

SteveMacenski, 2025-07-22:
You could create a ROS node or behavior tree node that would convert this into a nav-msgs/Path to be passed in for tracking. Right now there is not a provided way of doing this, but would be happy to have it added as a nav2_util + add the related BT node or ROS node as a utility function if generally useful for the user base to have in Nav2 - if you wanted to contribute it!

SteveMacenski, 2025-07-23:
There is no plan to integrate the limited constraints of VDA5050 onto Nav2. Frankly, I think that standard is lacking in modernity and expressiveness, but that's a story for another time. Suffice it to say I'm happy to support fleet management integration points, but I would not plan to limit those integrations via VDA5050. I'd want to support a much broader set of needs. 

I think what I suggested would work well. Have the curve be an input to the behavior tree or an independent ROS 2 node. Then, use a utility to convert it to some message that the standard servers can interact with and pass it off into them. Correct me if I'm wrong, but I don't think there's much (if any) downside to that & the NURBS->nav_msgs/Path shouldn't be a difficult conversion. That could be made as a `nav2_util` and then used as a BT Node to take and convert into a nav_msgs/Path for path tracking or what have you.

Calcifer852, 2025-07-24:
Thank you for your detailed response. I agree with your perspective. The goal is to support the VDA5050 standard with our new vehicle, hence opinions are out of the window. I appreciate your suggestion of converting the path to samples for use with the current Nav2 architecture. The only minor downside for my application is that, by sampling the curve, I’ll need to estimate the steering angle from the discretized path for tricycle model, even though this information is inherently available from the original curve. Otherwise, your proposed workflow makes sense, and I’ll proceed in that direction.

SteveMacenski, 2025-12-11:
Related for VDA 5050 https://github.com/ros-navigation/navigation2/pull/5362 https://github.com/ros-navigation/navigation2/issues/5037