ros-navigation/navigation2 #5410: Keepout zone unable to change dynamically

mini-1235, 2025-07-31:
## Bug report

#### Steps to reproduce issue

I am trying to dynamically change keepout zone by calling the service 
```
 ros2 service call /keepout_filter_mask_server/load_map nav2_msgs/srv/LoadMap "{map_url: '/root/nav2_ws/src/navigation2/nav2_bringup/maps/depot_keepout_dup.yaml'}"
```

after this service is called, there should be a /keepout_filter_mask received

the log also shows that we have received it 

```
[component_container_isolated-10] [INFO] [1753993363.636777201] [map_io]: Loading image_file: /root/nav2_ws/src/navigation2/nav2_bringup/maps/depot_keepout_dup.pgm
[component_container_isolated-10] [INFO] [1753993363.638780913] [map_io]: Read map /root/nav2_ws/src/navigation2/nav2_bringup/maps/depot_keepout_dup.pgm: 604 X 307 map @ 0.05 m/cell
[component_container_isolated-10] [WARN] [1753993363.639295950] [global_costmap.global_costmap]: KeepoutFilter: New filter mask arrived from /keepout_filter_mask topic. Updating old filter mask.
```

however, the map remain unchanged after I call the service

#### Reproduction instructions

1. Run tb4 demo

```
ros2 launch nav2_bringup tb4_simulation_launch.py headless:=0
```

2. call the service 
```
ros2 service call /keepout_filter_mask_server/load_map nav2_msgs/srv/LoadMap "{map_url: '/root/nav2_ws/src/navigation2/nav2_bringup/maps/depot_keepout_dup.yaml'}"
```

depot_keepout_dup.pgm is shown as follow

[image]

3. the keepout zone should be updated, however it remains unchanged

If I drive the robot near the keepout zone, it will partially change, although not entirely correct

[image]

#### Additional information

I believe this is related to this part of code
https://github.com/ros-navigation/navigation2/blob/c5531cb74ce5b1e06aba34e44fc21875f5cd0371/nav2_costmap_2d/src/layered_costmap.cpp#L158-L250
and I have managed to solve this problem by updating the min_x, min_y, max_x, max_y here
https://github.com/ros-navigation/navigation2/blob/c5531cb74ce5b1e06aba34e44fc21875f5cd0371/nav2_costmap_2d/src/layered_costmap.cpp#L186
which would involve adding a subscriber to /keepout_mask in costmap_filter.cpp

I am open to other suggestions

SteveMacenski, 2025-08-01:
What are the modifications you made? I'm curious, is the issue just that the visualization topic isn't updated? Is it actually otherwise using the new keepout zone? 

Also, check out https://github.com/ros-navigation/navigation2/pull/5187 which might be related. I have updates expand the bounds when necessary due to internal behaviors, maybe something like that makes sense here?

mini-1235, 2025-08-03:
>  I'm curious, is the issue just that the visualization topic isn't updated? Is it actually otherwise using the new keepout zone?

I don't think so, with the latest commit on rolling, I can see two results when I call the service:

1. When the robot does not move, calling the service does not work

[image]

2. When driving the robot near keepout zone, calling the service partially changed the keepout zone

[image]

With my code, even if not moving the robot, the keepout zone will switch to the new one

[image]

> Also, check out [#5187](https://github.com/ros-navigation/navigation2/pull/5187) which might be related. I have updates expand the bounds when necessary due to internal behaviors, maybe something like that makes sense here?

I think there are mainly 2 problems in the current code:

1.  We are having the wrong min and max here upon receiving the new map

https://github.com/ros-navigation/navigation2/blob/c5531cb74ce5b1e06aba34e44fc21875f5cd0371/nav2_costmap_2d/plugins/costmap_filters/keepout_filter.cpp#L160-L163

Static_layer actually has a variable `has_updated_data` which handles new map
https://github.com/ros-navigation/navigation2/blob/c5531cb74ce5b1e06aba34e44fc21875f5cd0371/nav2_costmap_2d/plugins/static_layer.cpp#L359-L376
I think this might be helpful in this issue

2. Keepout zone is only updating the value if data > old_data, if we want to handle this internally, then I think the condition here https://github.com/ros-navigation/navigation2/blob/c5531cb74ce5b1e06aba34e44fc21875f5cd0371/nav2_costmap_2d/plugins/costmap_filters/keepout_filter.cpp#L338 should be changed to 
```cpp
if (data > old_data || old_data == NO_INFORMATION || has_updated_data)
```

My current implementation updates the min and max when calling the `updateBounds` function in costmap filter, I have uploaded the code, can you check this commit
https://github.com/mini-1235/navigation2/commit/10b14f8ea9e34dda8d5052951ef3e9a9a00891fa

I think we can also handle map update internally in keepout_filter.cpp, let me know what you prefer

mini-1235, 2025-08-07:
> Another option would be to migrate the callback of the masks into the costmap filter base class to handle has_updated_data_ / other things the current implementation-specific costmap filter callbacks do. Then do what you have so that now there's only 1 mask callback (not 2) that can be given the topic name parameter for subscription. I didn't look into all the filters to see if there are any implementation consideration there, but at least looking at KeepoutFilter::maskCallback that's all trivially abstractable. That would be the best solution if that works and inline with your prototype.

Thanks for the suggestions @SteveMacenski , I spent time digging deeper into the code today, I think we can have `updateBounds` in Keepout filter directly, so we don't need to add a subscriber in costmap filter class. I think this is a cleaner and better solution compared to my prototype, I have raised a PR in https://github.com/ros-navigation/navigation2/pull/5429, let me know what do you think

mini-1235, 2025-08-07:
> Along these lines: a challenge I would provide you is to see if you can reimplement my logic in github.com/https://github.com/ros-navigation/navigation2/pull/5187 using your CostmapFilters base class / updateBounds method (for a follow up PR). I think your direction is better - though has the drawback of updating the bounds for all layers to update, not just the costmap filters or even just the keepout layer. It'll impact the sensor processing layers as well

Will do after this issue is closed!

closed 2025-08-08 by SteveMacenski