ros-navigation/navigation2 #5412: Request for adding a `current_pose` parameter / method to Simple Commander API of Nav2

Arseni1919, 2025-08-01:
# 🚀 Feature description

It would be awesome to add a `current_pose` method or property to Simple Commander API that returns the robot’s current `(x, y, yaw)` pose from `/amcl_pose` or localization source. Right now, you have to subscribe or parse topic output manually, which is a bit annoying.

Having a simple function like `nav.current_pose` would make scripting, debugging, and Jupyter notebooks way easier and cleaner! 🎉

💡 Implementation considerations
A straightforward way is to grab one message from` /amcl_pose` using something like:

```python
Copy
Edit
import subprocess, yaml, math

def get_amcl_pose():
    output = subprocess.check_output(["ros2", "topic", "echo", "--once", "/amcl_pose"], text=True)
    msg = list(yaml.safe_load_all(output))[0]
    p = msg['pose']['pose']['position']
    o = msg['pose']['pose']['orientation']
    yaw = math.atan2(2*(o['w']*o['z'] + o['x']*o['y']), 1 - 2*(o['y']**2 + o['z']**2))
    return p['x'], p['y'], yaw
```
Pros:

Super useful for quick checks and interactive use.

Easy to add, low overhead.

Could optionally support config for topic or TF fallback.

Thanks for considering! 🙌

SteveMacenski, 2025-08-01:
I think we should always use TF to check, not amcl_pose. That's there IMO as a debugging and state persistence topic and shouldn't be used as anything more than that. TF is for the time synchronized, accurate positioning data. 

It should be easy enough to add a TF call for map->base_link (or have them set the global and local frames) so that users can check on the frame - sure!

Arseni1919, 2025-08-02:
Oh, great! I will also change my function to look at the TF topic, as you suggested. Waiting for the update. Thank you in advance! 🙏

Arseni1919, 2025-08-14:
As far as I see it, the `/tf` topic does not contain the `map -> odom` frame transformation. Only `odom -> base_footprint` transformation in my case. I still need the into from `/amcl_pose` topic, right? I want the current position of my robot relative to the map.

SteveMacenski, 2025-08-14:
@claude use TF2 ROS to add to the Nav2 Simple Commander API package's SimpleNavigator a `getRobotPose(base_frame = 'base_link', map_frame='map')` method which uses TF 2 buffer and listener to obtain the transformation from map_frame to base_frame with default arguments "base_link" and "map" respectively. TF2 buffer operations can throw exceptions, so wrap it in a try/except and return None if the transform could not be completed with log warning. Else, return a PoseStamped of the transformation.

SteveMacenski, 2025-08-14:
@Arseni1919 checkout / test https://github.com/ros-navigation/navigation2/pull/5457

Arseni1919, 2025-09-20:
Great to see! Need to check it. Thank you for the help. And thanks to Claude Code 😉