ros-navigation/navigation2 #5419: nav2_route build error

Ericsii, 2025-08-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Computer:
  - ARM64
- ROS2 Version:
  - Humble apt binaries
- Version or commit hash:
  - humble_main(d50b68f0d)
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

I'm trying to build humble_main branch from source to use some nav2's new features in humble. Then I got following building error. It seems like something wrong with nanoflann.

```
-- Found ament_cmake: 1.3.12 (/opt/ros/humble/share/ament_cmake/cmake)
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found nav2_common: 1.4.0 (/root/T-REX/install/nav2_common/share/nav2_common/cmake)
-- Found rclcpp: 16.0.14 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.7 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.7 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.7 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.2 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.8 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclcpp_lifecycle: 16.0.14 (/opt/ros/humble/share/rclcpp_lifecycle/cmake)
-- Found rclcpp_components: 16.0.14 (/opt/ros/humble/share/rclcpp_components/cmake)
-- Found geometry_msgs: 4.9.0 (/opt/ros/humble/share/geometry_msgs/cmake)
-- Found nav2_costmap_2d: 1.4.0 (/root/T-REX/install/nav2_costmap_2d/share/nav2_costmap_2d/cmake)
-- Ensuring Eigen3 include directory is part of orocos-kdl CMake target
-- Found nav2_core: 1.4.0 (/root/T-REX/install/nav2_core/share/nav2_core/cmake)
-- Found visualization_msgs: 4.9.0 (/opt/ros/humble/share/visualization_msgs/cmake)
-- Found nlohmann_json: /usr/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.10.5") 
-- Setting build type to Release as none was specified.
-- Found ament_cmake_gtest: 1.3.12 (/opt/ros/humble/share/ament_cmake_gtest/cmake)
-- Found gtest sources under '/opt/ros/humble/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/T-REX/build/nav2_route
[  1%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/distance_scorer.cpp.o[0m
[  2%] [32mBuilding CXX object CMakeFiles/route_operations.dir/src/plugins/route_operations/adjust_speed_limit.cpp.o[0m
[  4%] [32mBuilding CXX object CMakeFiles/graph_file_loaders.dir/src/plugins/graph_file_loaders/geojson_graph_file_loader.cpp.o[0m
[  5%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/route_server.cpp.o[0m
[  7%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/route_planner.cpp.o[0m
[  8%] [32mBuilding CXX object gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o[0m
[ 10%] [32mBuilding CXX object CMakeFiles/graph_file_savers.dir/src/plugins/graph_file_savers/geojson_graph_file_saver.cpp.o[0m
[ 11%] [32mBuilding CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o[0m
[ 12%] [32m[1mLinking CXX static library libgtest_main.a[0m
[ 12%] Built target gtest_main
[ 14%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/route_tracker.cpp.o[0m
[ 15%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/time_scorer.cpp.o[0m
[ 17%] [32m[1mLinking CXX shared library libgraph_file_savers.so[0m
[ 17%] Built target graph_file_savers
[ 18%] [32mBuilding CXX object CMakeFiles/route_operations.dir/src/plugins/route_operations/trigger_event.cpp.o[0m
[ 20%] [32mBuilding CXX object CMakeFiles/route_operations.dir/src/plugins/route_operations/rerouting_service.cpp.o[0m
[ 21%] [32m[1mLinking CXX static library libgtest.a[0m
[ 22%] [32m[1mLinking CXX shared library libgraph_file_loaders.so[0m
[ 22%] Built target gtest
[ 24%] [32mBuilding CXX object CMakeFiles/route_operations.dir/src/plugins/route_operations/collision_monitor.cpp.o[0m
[ 24%] Built target graph_file_loaders
[ 25%] [32mBuilding CXX object CMakeFiles/route_operations.dir/src/plugins/route_operations/time_marker.cpp.o[0m
[ 27%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/edge_scorer.cpp.o[0m
[ 28%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/dynamic_edges_scorer.cpp.o[0m
[ 30%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/operations_manager.cpp.o[0m
[ 31%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/node_spatial_tree.cpp.o[0m
[ 32%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/path_converter.cpp.o[0m
[ 34%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/graph_loader.cpp.o[0m
[ 35%] [32m[1mLinking CXX shared library libroute_operations.so[0m
[ 35%] Built target route_operations
[ 37%] [32mBuilding CXX object CMakeFiles/route_server_core.dir/src/graph_saver.cpp.o[0m
[01m[K/root/T-REX/src/navigation2/nav2_route/src/node_spatial_tree.cpp:[m[K In member function ‘[01m[Kbool nav2_route::NodeSpatialTree::findNearestGraphNodesToPose(const PoseStamped&, std::vector<unsigned int>&)[m[K’:
[01m[K/root/T-REX/src/navigation2/nav2_route/src/node_spatial_tree.cpp:67:63:[m[K [01;31m[Kerror: [m[Kcannot convert ‘[01m[K__gnu_cxx::__alloc_traits<std::allocator<unsigned int>, unsigned int>::value_type*[m[K’ {aka ‘[01m[Kunsigned int*[m[K’} to ‘[01m[Klong unsigned int*[m[K’
   67 |   num_results = kdtree_->knnSearch(&query_pt[0], num_results, &ret_index[0], &out_dist_sqr[0]);
In file included from [01m[K/root/T-REX/src/navigation2/nav2_route/include/nav2_route/node_spatial_tree.hpp:21[m[K,
                 from [01m[K/root/T-REX/src/navigation2/nav2_route/src/node_spatial_tree.cpp:19[m[K:
[01m[K/usr/local/include/nanoflann.hpp:1253:31:[m[K [01;36m[Knote: [m[K  initializing argument 3 of ‘[01m[Ksize_t nanoflann::KDTreeSingleIndexAdaptor<Distance, DatasetAdaptor, DIM, IndexType>::knnSearch(const ElementType*, size_t, IndexType*, nanoflann::KDTreeSingleIndexAdaptor<Distance, DatasetAdaptor, DIM, IndexType>::DistanceType*, int) const [with Distance = nanoflann::L2_Simple_Adaptor<double, nav2_route::GraphAdaptor>; DatasetAdaptor = nav2_route::GraphAdaptor; int DIM = 2; IndexType = long unsigned int; size_t = long unsigned int; nanoflann::KDTreeSingleIndexAdaptor<Distance, DatasetAdaptor, DIM, IndexType>::ElementType = double; nanoflann::KDTreeSingleIndexAdaptor<Distance, DatasetAdaptor, DIM, IndexType>::DistanceType = double][m[K’
 1253 |                    [01;36m[KIndexType *out_indices[m[K, DistanceType *out_distances_sq,
      |                    [01;36m[K~~~~~~~~~~~^~~~~~~~~~~[m[K
gmake[2]: *** [CMakeFiles/route_server_core.dir/build.make:146: CMakeFiles/route_server_core.dir/src/node_spatial_tree.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[ 38%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/penalty_scorer.cpp.o[0m
[ 40%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/costmap_scorer.cpp.o[0m
[ 41%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/semantic_scorer.cpp.o[0m
[ 42%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/goal_orientation_scorer.cpp.o[0m
[ 44%] [32mBuilding CXX object CMakeFiles/edge_scorers.dir/src/plugins/edge_cost_functions/start_pose_orientation_scorer.cpp.o[0m
gmake[1]: *** [CMakeFiles/Makefile2:214: CMakeFiles/route_server_core.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[ 45%] [32m[1mLinking CXX shared library libedge_scorers.so[0m
[ 45%] Built target edge_scorers
gmake: *** [Makefile:146: all] Error 2

```

SteveMacenski, 2025-08-06:
I’ve built this recently on a Jetson Orin NX (ARM) fine so that is curious. Did you install Nanoflann manually or with rosdep? What compute platform, are you running in docker, & are you changing anything on the compiler or build flags side? 

I’m out for the night, but I’m alpha testing Claude Code to help with answering user Q&A and this seems like something it may do well. If it fumbles (which is probably will) lets keep chatting. 

@claude please respond to the user report. Note that this compiles fine on a Jetson Orin NX in a ROS Humble docker image, which is an ARM CPU

Ericsii, 2025-08-06:
I found in nanoflann the original function signature is 
```cpp
size_t KDTreeSingleIndexAdaptor::knnSearch(const ElementType *query_point, const size_t num_closest,
                   IndexType *out_indices, DistanceType *out_distances_sq,
                   const int /* nChecks_IGNORED */ = 10)
```
where `IndexType` is default to `size_t` aka `unsigned long` on ARM64 platform which is contradict to `node_spatial_tree.cpp:67`. I tried to change `node_spatial_tree.hpp:58` by setting the `IndexType` to `unsigned int` it fix the issue:

```cpp
typedef nanoflann::KDTreeSingleIndexAdaptor<
    nanoflann::L2_Simple_Adaptor<double, GraphAdaptor>, GraphAdaptor, DIMENSION, unsigned int> kd_tree_t;
```

Ericsii, 2025-08-06:
Should I create a PR?

commit bb532674 (2025-08-06): fix: resolve nanoflann type mismatch in nav2_route on ARM64

Fix compilation error on ARM64 Ubuntu 22.04 due to type mismatch 
between unsigned int* and size_t* when calling nanoflann's knnSearch.

- Change internal ret_index vector from std::vector<unsigned int> 
  to std::vector<size_t> for nanoflann compatibility
- Add static_cast<unsigned int> when populating output vector 
  to maintain API compatibility

Fixes #5419