ros-navigation/navigation2 #5442: Smac Planner: Collision Checker Optimization

SteveMacenski, 2025-08-11:
Right now, we pre-rotate the footprints by the orientation bins we know exist so that we have the polygons at those headings to remove the need to do trig ~a hundred thousand of times.

We can go one step further and instead of just storing the polygon's nodes at each orientation quantization, but store instead a vector of points from the line iterator at the orientations. Then, each iteration, we only need to lookup for that orientation quantization the vector of `mx,my` pairs that are offset from the origin (so just `mx+x`, `my+y`) and not have any additional computation other than checking the costmap itself. In other words, precompute those line iterators cells to check based on the robot being at the origin. Then we just have to add in the robot's actual position and we should have the costmap cell coordinates we need to check. 

I suspect this would yield some non-trivial performance improvements in this part of the code.

Adyansh04, 2025-08-24:
Hi, I would like to work on this.

SteveMacenski, 2025-08-27:
No one is working on this yet, this would be a great project! I think there's some other improvements we can make here as well possibly, but this is a good starting point :-)

SteveMacenski, 2025-12-05:
I would love improvements, but it doesn't seem this idea is that right idea, so closing. There's no requirement for more speed right now, but obviously would always love contributions if folks want / need it (or just a fun project)!