ros-navigation/navigation2 #5449: How should I interact with namespaced Nav2 TFs in Jazzy?

dvdmc, 2025-08-12:
I have been cracking my  head around setting up a multi-robot system with a Kobuki, robot_state_publisher, and Nav2 (the other robot is a drone using Aerostack2). As far as I understood, using a namespace in Nav2 will change the frames from /tf to tf. Then:
- How does it interact with i.e., robot state publisher?
- What should I modify in order to connect the tree between robot state publisher and Nav2?
- What if I have another robot (a drone) and would like to visualize all of them in RViz?

So far I have checked:
https://discourse.openrobotics.org/t/tf-tree-in-a-multi-robot-setup-in-ros2/41426
https://discourse.openrobotics.org/t/tf-tree-in-a-multi-robot-setup-in-ros2/41426
https://answers.ros.org/question/362879/
https://github.com/ros-navigation/navigation2/issues/2796 (wasn't able to solve this issue yet)
https://robotics.stackexchange.com/questions/102337/multi-robot-tf-with-namespaces-for-tf-or-with-frame-prefix-what-is-the-right-w
https://github.com/ros/robot_state_publisher/issues/125#issuecomment-577322986

But it's still unclear to me what is the intended way of simply namespacing TFs or how to interconnect things if there is TF tree isolation (which seem to be the Nav2 approach?). Also, is the global to local TF mapping used in Nav2 documented anywhere?

I know this might be quite an "open" question but I would really appreciate some light from the maintainers in this issue 🙏🏼

SteveMacenski, 2025-08-12:
Checkout `nav2_bringup` - we have examples with the TB3 demos that do this for N multirobot situations.

dvdmc, 2025-08-20:
I do have an answer to how to connect different namespaced topics, using remappings on node launch. Unfortunately, there is no equivalent for TF; however, for example, Aerostack2 implements its internal TF remapping strategy to prepend the node's namespace. That's why I am really interested in your take :) If you think it shouldn't be a Nav2 concern (then it neither should be AS2's), it should definitely be a TF concern, even if the solution is to create a node to subscribe and bridge different namespaced TFs. Thanks for your insights and the discussion! I'll let you know about my solution in case it's useful for Nav2.