ros-navigation/navigation2 #5456: FollowPath BT plugin input port wrong in nav2_tree_nodes.xml on Jazzy

DavidG-Develop, 2025-08-14:
## Bug report

The FollowPath Action plugin in nav2_tree_nodes.xml on jazzy is defined to have <input_port name="service_name">Service name</input_port> instead of <input_port name="server_name">**Server name**</input_port>. As the FollowPath Action plugin is a child class of BtActionNode and not BtServiceNode the correct input port is server_name.

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- ROS2 Version:
  - Jazzy binary
- Version or commit hash:
  - ros-jazzy-navigation2	1.3.6-1noble.20250520.223628
- DDS implementation:
  - Cyclone dds

#### Steps to reproduce issue

Load up Groot -> load BT you want to use -> change service_name in FollowPath action -> Save -> Clean up empty strings left by Groot 🙃 -> boot up nav2

#### Expected behavior

To work :) .

#### Actual behavior

Does not work :( . 

[dumper.nav.bt_navigator]: Exception when loading BT: a port with name [service_name] is found in the XML (<FollowPath>, line 20) but not in the providedPorts() of its registered node type.

mini-1235, 2025-08-14:
This is fixed in main in https://github.com/ros-navigation/navigation2/pull/4931, I think we can backport that

DavidG-Develop, 2025-08-14:
Thanks! If the backport runs into any issues, I'm happy to help with a PR for this issue.

SteveMacenski, 2025-08-14:
@DavidG-Develop can you open a backport of just that portion of the PR? I can't backport the entire PR