ros-navigation/navigation2 #5465: MPPI weird behavior when ax_min very different from ax_max

tonynajjar, 2025-08-19:
## Bug report

**Required Info:**

- ROS2 Version:
  - Jazzy
- Version or commit hash:
  - 6206406d20daed6f595f8aa3723476eae776e70f + custom changes (likely unrelated to this bug)

#### Steps to reproduce issue

For my use case I want to decelerate much faster then I accelerate e.g `ax_max: 0.2` and `ax_min: -2.0`. I assume that's a valid configuration no?

#### Expected behavior

The optimal and candidate trajectories become very unstable - not sure what's happening internally

#### Actual behavior

Stable trajectories and simple acceleration limiting as if done by the velocity_smoother

#### Reproduction instructions

#### Additional information

SteveMacenski, 2025-08-19:
A couple of question:
- Related to https://github.com/ros-navigation/navigation2/issues/5464? 
- If you set it to -0.2 - 0.2 or -2.0 - 2.0 do you see an issue? Just making sure its not an issue with fast or slow accelerations themselves and it has to do with asymmetry. Note that the PR #5464 takes issue with was in response to supporting slower acceleration profiles in a more stable fashion if you read its original ticket thread 
- What is the behavior that you see and is this using the motion model changes from 5464?

SteveMacenski, 2026-04-09:
https://github.com/ros-navigation/navigation2/tree/low_accel resolves

closed 2026-04-09 by SteveMacenski