ros-navigation/navigation2 #5470: AMCL intermittent crash on Raspberry Pi 5 with stack smashing detected

EricoMeger, 2025-08-20:
- Operating System:
  - Ubuntu 24.04
- Computer:
  - Raspberry Pi 5 8 gb
- ROS2 Version:
  - Jazzy installed via deb packages
- Version or commit hash:
  - ros-jazzy-navigation2	1.3.7-1noble.20250624.214056
  - ros-jazzy-nav2-amcl	1.3.7-1noble.20250624.150636
- DDS implementation:
  -  Eclipse Zenoh

#### Steps to reproduce issue
I cannot reliably reproduce the crash; it happens intermittently during startup of the navigation stack with AMCL. 

Typical steps that sometimes trigger it:

1. Launch the full navigation stack:
   ```bash
   ros2 launch nav_pkg nav.launch.py

#### Expected behavior
AMCL should start and configure correctly

#### Actual behavior
Occasionally, AMCL crashes during startup with a stack smashing detected error at `initTransforms`. The crash happens intermittently and is not reliably reproducible.

```
[map_server-1] [INFO] [1755174650.889985773] [map_io]: Read map /home/ubuntu/ros2_ws/src/nav_hub/maps/mapa360.pgm: 782 X 1167 map @ 0.036 m/cell
[lifecycle_manager-10] [INFO] [1755174650.901656774] [lifecycle_manager_navigation]: Configuring amcl
[amcl-3] [INFO] [1755174650.902333369] [amcl]: Configuring
[amcl-3] [INFO] [1755174650.902474739] [amcl]: initTransforms
[amcl-3] *** stack smashing detected ***: terminated
```

Error on backtrace:

```
#0  __pthread_kill_implementation (threadid=281474840551456, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x0000fffff76f7670 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000fffff76acb3c in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000fffff7697e00 in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000fffff76eabf4 in __libc_message_impl (fmt=fmt@entry=0xfffff77d2b18 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:134
#5  0x0000fffff77681a8 in __GI___fortify_fail (msg=msg@entry=0xfffff77d2b30 "stack smashing detected") at ./debug/fortify_fail.c:24
#6  0x0000fffff7769164 in __stack_chk_fail () at ./debug/stack_chk_fail.c:24
#7  0x0000fffff6f7338c in eigen_decomposition () from /opt/ros/jazzy/lib/libpf_lib.so
#8  0x0000fffff6f73410 in pf_matrix_unitary () from /opt/ros/jazzy/lib/libpf_lib.so
#9  0x0000fffff6f7353c in pf_pdf_gaussian_alloc () from /opt/ros/jazzy/lib/libpf_lib.so
#10 0x0000fffff6f73668 in pf_init () from /opt/ros/jazzy/lib/libpf_lib.so
#11 0x0000fffff7eddb44 in nav2_amcl::AmclNode::initParticleFilter() () from /opt/ros/jazzy/lib/libamcl_core.so
#12 0x0000fffff7ece6fc in nav2_amcl::AmclNode::on_configure(rclcpp_lifecycle::State const&) () from /opt/ros/jazzy/lib/libamcl_core.so
```

#### Reproduction instructions
Cannot reliably reproduce. Typical usage that sometimes triggers the crash:

 ```
   ros2 launch nav_pkg nav.launch.py
```

Wait for AMCL to configure and initialize. The crash may appear in the console and GDB backtrace shows the failure in `eigen_decomposition()` inside `libpf_lib.so`.

#### Additional information
- The crash occurs intermittently: approximately 2 out of every 5 launches trigger it. However, it is not consistent, and sometimes it does not occur at all.
- I'm using a LDROBOT STL27L LiDAR publishing ~2170 points per scan with some NaN values in ranges and intensity
- I tried tweaking some AMCL parameters: reducing max_beams, max_particles, and min_particles, but the crash persisted. I also considered that the issue might be related to the initial pose, so I tried setting set_initial_pose and always_reset_initial_pose to true, but that did not help.
- I also suspected the map could be related, and tried creating a new one and testing maps known to work fine on Humble, but the problem persisted.
- htop memory and CPU usage at the moment of the crash: [Watch Here](https://youtu.be/zTtWepKJXAk)
<details>
<summary> AMCL parameters</summary>
[... 610 lines cut ...]
                            {'autostart': autostart},
                            {'node_names': lifecycle_nodes}]),
        ]
    )

    # Create the launch description and populate
    ld = LaunchDescription()

    # Set environment variables
    ld.add_action(stdout_linebuf_envvar)

    # Declare the launch options
    ld.add_action(declare_namespace_cmd)
    ld.add_action(declare_use_sim_time_cmd)
    ld.add_action(declare_params_file_cmd)
    ld.add_action(declare_autostart_cmd)
    ld.add_action(declare_container_name_cmd)
    ld.add_action(declare_use_respawn_cmd)
    ld.add_action(declare_log_level_cmd)
    # Add the actions to launch all of the navigation nodes
    ld.add_action(load_nodes)

    return ld
```
</details>

EricoMeger, 2025-08-21:
I had a couple of issues trying to build Nav2 with the Debug option to get a more detailed backtrace, and the build time on the RPi is quite long. I will be able to provide the full backtrace tomorrow, sorry for the delay.

For now, I’ll go ahead and answer your questions.

Also, even though there isn’t much additional information, I’m attaching the entire run logs instead of just a cherry-picked section:

<details>
<summary>Run logs</summary>

```yaml
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-08-21-09-44-20-841266-robot06-4564
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [4567]
[INFO] [controller_server-2]: process started with pid [4568]
[INFO] [amcl-3]: process started with pid [4569]
[INFO] [smoother_server-4]: process started with pid [4570]
[INFO] [planner_server-5]: process started with pid [4571]
[INFO] [behavior_server-6]: process started with pid [4572]
[INFO] [bt_navigator-7]: process started with pid [4573]
[INFO] [waypoint_follower-8]: process started with pid [4574]
[INFO] [velocity_smoother-9]: process started with pid [4575]
[INFO] [lifecycle_manager-10]: process started with pid [4576]
[velocity_smoother-9] [INFO] [1755780261.259559858] [velocity_smoother]: 
[velocity_smoother-9] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-9] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[behavior_server-6] [INFO] [1755780261.285883982] [behavior_server]: 
[behavior_server-6] 	behavior_server lifecycle node launched. 
[behavior_server-6] 	Waiting on external lifecycle transitions to activate
[behavior_server-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-2] [INFO] [1755780261.345489609] [controller_server]: 
[controller_server-2] 	controller_server lifecycle node launched. 
[controller_server-2] 	Waiting on external lifecycle transitions to activate
[controller_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-2] [INFO] [1755780261.355782193] [controller_server]: Creating controller server
[lifecycle_manager-10] [INFO] [1755780261.364507777] [lifecycle_manager_navigation]: Creating
[controller_server-2] [INFO] [1755780261.366198816] [local_costmap.local_costmap]: 
[controller_server-2] 	local_costmap lifecycle node launched. 
[controller_server-2] 	Waiting on external lifecycle transitions to activate
[controller_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-2] [INFO] [1755780261.367238272] [local_costmap.local_costmap]: Creating Costmap
[lifecycle_manager-10] [INFO] [1755780261.384470679] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[smoother_server-4] [INFO] [1755780261.390330266] [smoother_server]: 
[smoother_server-4] 	smoother_server lifecycle node launched. 
[smoother_server-4] 	Waiting on external lifecycle transitions to activate
[smoother_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-4] [INFO] [1755780261.394113959] [smoother_server]: Creating smoother server
[bt_navigator-7] [INFO] [1755780261.402307592] [bt_navigator]: 
[bt_navigator-7] 	bt_navigator lifecycle node launched. 
[bt_navigator-7] 	Waiting on external lifecycle transitions to activate
[bt_navigator-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-7] [INFO] [1755780261.405373443] [bt_navigator]: Creating
[map_server-1] [INFO] [1755780261.421808432] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1755780261.422022861] [map_server]: Creating
[waypoint_follower-8] [INFO] [1755780261.432515762] [waypoint_follower]: 
[waypoint_follower-8] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-8] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-8] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-8] [INFO] [1755780261.433490533] [waypoint_follower]: Creating
[planner_server-5] [INFO] [1755780261.439060543] [planner_server]: 
[planner_server-5] 	planner_server lifecycle node launched. 
[planner_server-5] 	Waiting on external lifecycle transitions to activate
[planner_server-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-5] [INFO] [1755780261.443915063] [planner_server]: Creating
[planner_server-5] [INFO] [1755780261.450971072] [global_costmap.global_costmap]: 
[planner_server-5] 	global_costmap lifecycle node launched. 
[planner_server-5] 	Waiting on external lifecycle transitions to activate
[... 51 lines cut ...]
>  Can you reproduce with Nav2 default config? I’m interested if it’s configuration related.

I wasn’t sure if you meant literally running with no custom configuration, but I tested with the `sam_bot` nav_params from the Nav2 tutorials, and the crash still occurred.

[... cut ...]
From what I can see, there aren’t major differences in configs between Humble and Jazzy. However, while on Humble we were running on a Raspberry Pi 4. If you think it’s worthwhile, I could try running a container with Ubuntu 22.04 + Humble on the RPi 5 (I can’t install it natively).

> Does this reproduce in another computer? Wanting to know if it’s a RPi specific issue.

No, I couldn’t reproduce the issue on another computer, so I’m inclined to believe it may be RPi-specific.

Yesterday I ran our robot on a notebook (i7-8565U, 8GB RAM, Ubuntu 24.04 + Jazzy). In more than 20 runs, AMCL did not crash once (same map, configs, etc).

To be sure, I’ll try testing on another computer later.

SteveMacenski, 2025-09-02:
I think this line says the important part:

```
#11 0x0000fffff7eae014 in nav2_amcl::AmclNode::initParticleFilter (this=this@entry=0xaaaaaacd45b0) at /home/ubuntu/ros2_ws/src/navigation2/nav2_amcl/src/amcl_node.cpp:1633
    pf_init_pose_mean = {v = {-nan(0xfffffff010000), 1.4853970540215366e-313, 9.8813129168249309e-324}}
    pf_init_pose_cov = {m = {{9.2711411887811341e-310, 0, 0}, {0, 1.2055201758526416e-321, 0}, {0, 0, -nan(0xfffffff010000)}}}

```

This means that you have NaNs coming in from the initial pose and covariance. It makes sense that you don't have symbols for everything since you didn't compile rclcpp with debug flags and that's A-OK. The issue is NaNs in your input.

EricoMeger, 2025-09-04:
Sorry for taking up your time, and thanks for your responses so far. Could you clarify where AMCL should be getting the initial pose and covariance from?

The backtrace I shared was generated when running AMCL without setting any initial pose in the config, and it didn’t even start for me to be able to set a pose through RViz.

Even when I explicitly set an initial pose in the config with default values like:

```bash
set_initial_pose: true 
initial_pose: 
    x: 0.0 
    y: 0.0 
    z: 0.0 
    yaw: 0.0
```

I still occasionally see this error.

One thing I realized after discussing with my lab partners is that this issue only started happening when we switched our DDS to Zenoh. Back when we were using Cyclone, we didn’t encounter this problem. Do you think that could be related?

SteveMacenski, 2025-09-04:
> One thing I realized after discussing with my lab partners is that this issue only started happening when we switched our DDS to Zenoh. Back when we were using Cyclone, we didn’t encounter this problem. Do you think that could be related?

I do not know. Can you switch back to Cyclone to check?

SteveMacenski, 2025-09-26:
I doubt its launching issue, but when you have it in the default Nav2 launch files then things are being loaded into a component container `nav2_container` into the same process. You could reproduce this with `use_composition` as `false` on bringup to launch each node in its own process rather than into a single process (which comes with runtime computation / latency overhead). 

I don't think that doesn't mean its an issue with RMW Zenoh, it could mean that its not well handling composition? If we can't reproduce with another middleware, that makes me think its not rclcpp/rcl/rmw - bu rather rmw zenoh. I think what you found is just another good signal which can help isolate and identify the issue :-)