ros-navigation/navigation2 #5524: Open Loop vs Closed Loop in Controller Server, Velocity Smoother

mini-1235, 2025-09-15:
I was reading about DWPP https://github.com/Decwest/nav2_dynamic_window_pure_pursuit_controller, which shows that it has better performance than RPP. I noticed that it has an option to choose open loop/closed loop in the plugin, https://github.com/Decwest/nav2_dynamic_window_pure_pursuit_controller/blob/bf281cd92d74d748794aea386f767f9c8d67f78f/src/dynamic_window_pure_pursuit_controller.cpp#L257-L264

Currently in the codebase, we have only closed loop mode in controller server, except for rotation shim, which was introduced in https://github.com/ros-navigation/navigation2/pull/4880. For velocity smoother, we have closed loop / open loop(default), which means by default, we will have:

- Controller Server (closed loop)
- Velocity Smoother (open loop)

From the description of velocity smoother's parameters:
> Type of feedback to use for the current state of the robot’s velocity. In OPEN_LOOP, it will use the last commanded velocity as the next iteration’s current velocity. When acceleration limits are set appropriately, this is a good assumption. In CLOSED_LOOP, it will use the odometry from the odom topic to estimate the robot’s current speed. In closed loop mode, it is important that the odometry is high rate and low latency, relative to the smoothing frequency.

I am actually curious about what happen if we set it to open loop + open loop, for example when we have set the acceleration limits appropriately, or when we do not have high rate odometry. Will the result be better theoretically?

I have actually tried closed loop + closed loop (I have a high rate odometry) a few weeks ago, but the result is worse than closed loop + open loop, I guess I didn't set the limits appropriately

mini-1235, 2025-09-16:
> So the open / closed loop in the Velocity Smoother comes from my research and consolidation of features across Velocity Smoothers from ROS 1. I was aware of a half dozen or so Velocity Smoothers that each worked slightly differently, with slightly different behavior, for slightly different applications - but broadly interchangeable. I put together a feature set by reviewing all implementations I could find and developing a package for Nav2 that incorporated all of the implementations' features in one concise package so that there wasn't a need to create additional ones in the future with the goal of eliminating the need of additional Velocity Smoothers. Closed / Open Loop were one of those features and useful in the context of highly delayed or particularly low-quality odometry.

I tested the closed loop mode of velocity smoother on tb3/tb4, and noticed that the robots occasionally get stuck on obstacles. I assume the default configuration already applies the proper limits, and /odom runs at 30 Hz compared to the velocity smoother’s 20 Hz frequency. Do you have any thoughts on why closed loop mode velocity smoother might be problematic for tb3/tb4? Also, do we have any examples of velocity smoother configured in closed loop mode so that I can take a look?

> Certain algorithms like RPP and Graceful really don't use the input velocity so its moot. Those are more for DWB and MPPI which consider the dynamic limitations from the current speed when calculating the next trajectory. The speed in RPP is only used to find the lookahead carrot distance when using adaptive lookahead distances and for rotating to heading - but not the actual path following.

I am actually interested in the rotate to heading part as #4880 introduces open loop mode to the rotation shim controller, RPP has the exact same logic for calculating wz. Is that PR intended to solve cases where the rotation shim is used with low-rate odometry, such that the current speed estimate is not reliable?

SteveMacenski, 2025-09-16:
>  and /odom runs at 30 Hz compared to the velocity smoother’s 20 Hz frequency. Do you have any thoughts on why closed loop mode velocity smoother might be problematic for tb3/tb4?

What does 'stuck on obstacles' mean? Also, odom for real hardware robots is typically at about 100hz or higher, so try increasing that in gazebo. I'm not sure if there's any simulator downside to that or not. 30hz was just taken from Robotis / Clearpath on their simulators. 

> Is that PR intended to solve cases where the rotation shim is used with low-rate odometry, such that the current speed estimate is not reliable?

See https://github.com/ros-navigation/docs.nav2.org/pull/639/files

mini-1235, 2025-09-17:
I tried to do some experiments on tb4 today to see how different odometry rates and feedback types affect the controller. From visualizing the robot in RViz and looking at the /cmd_vel plots, I think it is hard to clearly show my result here. Are there any tools I can use to more clearly demonstrate the performance actually improves after a change? I noticed that we have a benchmark tool in MPPI, is that something I can use here?

Decwest, 2025-09-18:
Sorry for my late reply, and thank you very much for mentioning me.

**Regarding the integration of DWPP** @SteveMacenski

I strongly wish for this integration. In fact, I had been planning to reach out myself, so I am truly delighted and honored that you brought this up.
I have sent the PDF of the paper to the email address listed on your GitHub profile. I would be grateful if you could kindly take a look.

The advantages of introducing DWPP for robot behavior are as follows:
By computing the velocity that follows the path optimally under velocity and acceleration constraints, the tracking error is reduced. This effect is especially significant on paths with large curvature, where the method generates smoother behaviors with less overshoot, by appropriately reducing linear velocity while turning.

**Regarding OPEN_LOOP / CLOSED_LOOP**
Since there is inevitably a delay between when the velocity command is published and when the robot’s joints rotate to reflect that velocity, I think that in CLOSED_LOOP mode there may be cases where proper acceleration and deceleration cannot be achieved.
For proper acceleration and deceleration, it is necessary that the previous velocity command has been realized and reflected in the /odom when the next command is computed. Therefore, I suspect that the higher the publish frequency of the velocity command, the more difficult it may become to achieve appropriate acceleration and deceleration.

Decwest, 2025-09-20:
Thank you for reading my paper. Your understanding is perfectly correct.
Regarding proactively slowing down before a sharp turn, I described a concrete example where DWPP slows down before the sharp turn while conventional PPs do not:
https://github.com/Decwest/nav2_dynamic_window_pure_pursuit_controller/blob/main/algorithm.md#:~:text=Difference%20of%20actual%20velocity%20realized%20by%20conventional%20PP%20variants%20and%20DWPP

Of course, let me implement it and create a PR!

To start, should I begin with Humble support by branching off from the [humble](https://github.com/ros-navigation/navigation2/tree/humble) branch and then create a PR targeting the humble branch? (Support for other versions can be added easily.)