ros-navigation/navigation2 #5531: [MPPI] wobbling issue on real robot

ironoa, 2025-09-17:
Hello,
I am facing a wobbling issue with my Differential Drive robot. I am trying to use the MPPI controller.

**Required Info:**

- Operating System:
  - Ubuntu 24.04 inside Docker
- ROS2 Version:
  - Jazzy (Dockerized)
- Nav2 Package:
  - MPPI (nav2_mppi_controller)
- Have you asked this on Robotics Stack Exchange?
  - No

#### Tuning / Configuration Goal
Achieve smooth, straight tracking on a differential-drive robot in the real world (casters on the front), comparable to simulation, without the wobble/heading overshoot. In simulation the same config is rock solid; on hardware it wobbles, not only at the start. With RPP / Pure Pursuit (lookahead 1.0 m) the robot is rock solid on hardware, that's currently my plan B (although I am trying to avoid it).
The end goal is a coverage task, but here I am really talking about sending a single pose with Navigate To Pose, or a few of them with Navigate Though Poses

#### What I’m Observing
- Simulation: straight, stable behavior.
- Real robot: wobbling, heading overshoots on straight segments. 
- Hypothesis: front caster wheels induce the robot to steer a bit. With the radio controller, I could just go straight and ignore that minor steering, that would settle quickly anyway; MPPI may be “chasing” it instead, making it fall down a spiral. 

#### Configuration Snippet
```
controller_server:
  ros__parameters:
    enable_stamped_cmd_vel: true
    use_sim_time: false
    controller_frequency: 30.0 
    progress_checker_plugin: "progress_checker"
    goal_checker_plugins: ["goal_checker"]
    controller_plugins: ["FollowPath"]
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.1
      movement_time_allowance: 30.0
    goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      stateful: true
      xy_goal_tolerance: 0.5
      yaw_goal_tolerance: 0.5
    FollowPath: #Default controller
      plugin: "nav2_mppi_controller::MPPIController" #https://docs.nav2.org/configuration/packages/configuring-mppic.html
      time_steps: 56
      model_dt: 0.05 # => prediction horizon 56*0.05=2.8 | threshold_to_consider ~= prediction horizon * vx_max, for instance 1.4 when v is 0.5
      batch_size: 2000
      vx_std: 0.2
      vy_std: 0.0
      wz_std: 0.2
      vx_max: 0.5
      vx_min: -0.35
      vy_max: 0.0
      wz_max: 1.9
      ax_max: 2.0
      ax_min: -2.0
      az_max: 2.5
      iteration_count: 1
      prune_distance: 1.7
      transform_tolerance: 0.1
      temperature: 0.3
      motion_model: "DiffDrive"
      visualize: false
      TrajectoryVisualizer:
        trajectory_step: 5
        time_step: 3
      critics: ["ConstraintCritic", "CostCritic", "GoalCritic", "GoalAngleCritic", "PathAlignCritic", "PathFollowCritic", "PathAngleCritic", "PreferForwardCritic"]
      ConstraintCritic:
        enabled: true
        cost_power: 1
        cost_weight: 4.0
      GoalCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
      GoalAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 3.0
      PreferForwardCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
      ObstaclesCritic:
        enabled: false
        cost_power: 1
        repulsion_weight: 3.0
        critical_weight: 25.0
        consider_footprint: true
        collision_cost: 10000.0
        collision_margin_distance: 4.0
        near_goal_distance: 0.3
      CostCritic:
        enabled: false
        consider_footprint: true
      PathAlignCritic:
        enabled: true
        cost_power: 1
        cost_weight: 10.0
      PathFollowCritic:
        enabled: true
        cost_power: 1
        cost_weight: 5.0
      PathAngleCritic:
        enabled: true
        cost_power: 1
        cost_weight: 2.2
        # forward_preference: true
        mode: 0 #0 is forward preference
```

#### What I've Tried
- Changing accel limits: Increasing ax_max relative to az_max reduces wobble somewhat, but not fully
- Making any of the 3 Path critics be prevalent respect to the others
- Comparison with RPP: Using RPP / Pure Pursuit with 1.0 m lookahead is stable on the same robot and environment
- Docs suggest tuning Obstacle vs PathFollow with inflation to avoid “chasing zero-cost pockets.” I actually disabled Obstacle critic completely. I am literally trying to go from point A to B in a straight line
- (I haven’t yet rebalanced critics or changed sampling; happy to test specific suggestions.)

Thanks for any guidance on best-practice tuning

ironoa, 2025-09-19:
I'll get back with a rviz capture to better show the problem...

>> Is the hardware compute the same? Do you see any warnings about missed control loop rates?

No warnings in the logs. I tried with MPPI controller_frequency rates 20 (matching my diff_drive_controller publish_rate) and 30 (the default on the [documentation](https://docs.nav2.org/configuration/packages/configuring-mppic.html)) with no noticeable difference.

Tested on different platforms, though with htop there were no signs of struggling in the Real Robot resources:

Simulation:
- beefy server
- ryzen 9950x3d 
- 16 GB ram (allocated in a VM)
- rtx 2060
- vm with ubuntu 24.04 -> 16GB ram, 8 cores, smt disabled
- Docker

Simulation:
- macbook pro m1 (arm)
- vm with ubuntu 24.04 -> 16GB ram, 4 cores
- Docker

Simulation:
- macbook pro intel i9
- vm with ubuntu 24.04 -> 16GB ram, 4 cores
- Docker

Real Robot:
- 4 cores, 8 threads 
- 16 GB ram
- htop shows no problems during the run; plenty of ram available, cores under 50% usage
- Docker

>> Do you see any changes with / without any critics?

Nothing that I can really appreciate, but probably it's also because of my lack of experience... not sure which methodological direction to take. And not sure if this is an expected "initial tuning" behavior everybody need to face at the beginning, or if there is something fundamentally wrong on my side

sualeh-rgt, 2025-09-19:
you are not alone in this!
running jazzy on ubuntu 24.04.

In simulation works near perfect, but on real robot with exact same configuration velocity stays 10 times lower than expected. 
occasionally spins fast and then continues to drive in low speed. 
 
1. cmd_vel from the controller stays low 
2. odom shows the current speed with slight difference
3. keyboad control drives robot with correct speed.
 
tweeked costmap, obstacle sources but no difference.

ironoa, 2025-10-24:
> [@ironoa](https://github.com/ironoa) any update?

Sorry @SteveMacenski, not working on this anymore. If you agree I could close the issue and let others eventually reference it

closed 2025-11-18 by SteveMacenski