ros-navigation/navigation2 #5538: Route Server missing from apt repo

Manouselis, 2025-09-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 
- Computer:
  - 13th Gen Intel i9-13950HX
- ROS2 Version:
  -  Humble Binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.18-1jammy.20250728.103122
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
sudo apt update & sudo apt upgrade
ros2 pkg list | grep route
```

#### Expected behavior
Showing the route server package 

#### Actual behavior
Not showing anything

#### Reproduction instructions

#### Additional information

closed 2025-09-19 by SteveMacenski

mini-1235, 2025-11-14:
> Hello, it displayed: E: Package not found ros-humble-nav2-route I have Humble and Jazzy, both of which cannot get the package.

Did you run `apt update`? On jazzy, I can find the packages successfully:
```
> apt search nav2-route
Sorting... Done
Full Text Search... Done
ros-jazzy-nav2-route/noble 1.3.9-1noble.20251008.072146 amd64
  A Route Graph planner to compliment the Planner Server
```

mini-1235, 2025-11-15:
> Thank you for your confirmation. I also can't find the package in the jazzy version.

Can you double check your apt sources? I have just pulled a fresh image here and the package appears correctly.
Nav2 also provides docker images https://github.com/ros-navigation/nav2_docker, maybe that works for you?

SteveMacenski, 2025-11-15:
I pulled the most recent Jazzy container, did an apt update/upgrade and rosdep update, then I see it when doing `apt install ros-<distro>-nav2-route`. Humble I agree I'm seeing an issue. The binary build job is failing https://build.ros2.org/job/Hbin_uJ64__nav2_route__ubuntu_jammy_amd64__binary/65/consoleFull that I missed the emails on.

```
23:40:18 /usr/bin/ld: /tmp/ccqYIXIS.ltrans0.ltrans.o: in function `RouteServerTest_test_minor_utils_Test::TestBody()':
23:40:18 ./.obj-x86_64-linux-gnu/test/./test/test_route_server.cpp:123: undefined reference to `void nav2_route::RouteServer::exceptionWarning<nav2_msgs::action::ComputeRoute_Goal_<std::allocator<void> > >(std::shared_ptr<nav2_msgs::action::ComputeRoute_Goal_<std::allocator<void> > const>, std::exception const&)'
23:40:18 /usr/bin/ld: /tmp/ccqYIXIS.ltrans0.ltrans.o: in function `RouteServerTest_test_request_valid_Test::TestBody()':
23:40:18 ./.obj-x86_64-linux-gnu/test/./test/test_route_server.cpp:142: undefined reference to `bool nav2_route::RouteServer::isRequestValid<nav2_msgs::action::ComputeAndTrackRoute>(std::shared_ptr<nav2_util::SimpleActionServer<nav2_msgs::action::ComputeAndTrackRoute> >&)'
23:40:18 /usr/bin/ld: ./.obj-x86_64-linux-gnu/test/./test/test_route_server.cpp:142: undefined reference to `bool nav2_route::RouteServer::isRequestValid<nav2_msgs::action::ComputeAndTrackRoute>(std::shared_ptr<nav2_util::SimpleActionServer<nav2_msgs::action::ComputeAndTrackRoute> >&)'
23:40:18 /usr/bin/ld: ./.obj-x86_64-linux-gnu/test/./test/test_route_server.cpp:142: undefined reference to `bool nav2_route::RouteServer::isRequestValid<nav2_msgs::action::ComputeAndTrackRoute>(std::shared_ptr<nav2_util::SimpleActionServer<nav2_msgs::action::ComputeAndTrackRoute> >&)'
23:40:18 /usr/bin/ld: ./.obj-x86_64-linux-gnu/test/./test/test_route_server.cpp:142: undefined reference to `bool nav2_route::RouteServer::isRequestValid<nav2_msgs::action::ComputeAndTrackRoute>(std::shared_ptr<nav2_util::SimpleActionServer<nav2_msgs::action::ComputeAndTrackRoute> >&)'
23:40:18 collect2: error: ld returned 1 exit status
```

This is a neat problem. I don't see this in CI or my local builds. I'm pretty sure the issue is that we need to move the templated methods into the header or add in explicit template initializations. Given this is only an issue for humble, I lean towards the template initialization direction to make these files as consistent as possible with more up to date distributions (and we do this anyway in the goal intent extractors). I hate, as much as possible, to have implementation logic in the headers -- except when templated by many, many types. 

See https://github.com/ros-navigation/navigation2/pull/5686. After a re-release this should work. In the meantime, compile from source unfortunately is the best answer to use this feature. It should compile and work fine :-)

iumida, 2025-11-18:
I've confirmed again that jazzy has found nav2_route. Thank you.