ros-navigation/navigation2 #5549: BT: Blackboard::get() error

Wibaz, 2025-09-24:
**Required Info:**

- Operating System:
  - Ubuntu 22.04 (Jammy) [TODO: confirm]
- Computer:
  - [TODO: e.g., Nvidia Jetson Orin, 13th Gen Intel NUC, Ryzen 9 7940HS]
- ROS2 Version:
  - Humble (apt binaries)
- Version or commit hash:
  - apt binaries (please see exact versions):
    - Output of: `dpkg-query --show "ros-$ROS_DISTRO-navigation2"`
    - Output of: `dpkg-query --show "ros-$ROS_DISTRO-nav2-*"`
- DDS implementation:
  - [TODO: e.g., rmw_fastrtps_cpp (default), CycloneDDS, etc.]
- Nav2 Package:
  - nav2_bt_navigator (Behavior Tree Navigator)
- Have you asked this on Robotics Stack Exchange?
  - No

#### Tuning / Configuration Goal
Bring up Nav2 and use the default Navigate Through Poses BT with replanning and recovery on ROS 2 Humble. The BT Navigator fails to activate due to a missing blackboard key required by the XML.

#### What I've Tried
- Launching Nav2 bringup (component_container_isolated) with default configs.
- Using the stock behavior tree XML installed by Humble:
  - `/opt/ros/humble/share/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml`
- On activation, `bt_navigator` throws a blackboard error indicating a missing key `wait_for_service_timeout` and fails to load the XML. As a result, lifecycle bringup aborts.

- Launch logs:
  ```
  [component_container_isolated-3] [INFO] [1758713047.276377347] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
  [component_container_isolated-3] [INFO] [1758713047.276405143] [lifecycle_manager_navigation]: Activating bt_navigator
  [component_container_isolated-3] [INFO] [1758713047.276611253] [bt_navigator]: Activating
  [component_container_isolated-3] [ERROR] [1758713047.276898902] [bt_navigator]: Exception when loading BT: Blackboard::get() error. Missing key [wait_for_service_timeout]
  [component_container_isolated-3] [ERROR] [1758713047.276916374] [bt_navigator]: Error loading XML file: /opt/ros/humble/share/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml
  [component_container_isolated-3] [ERROR] [1758713047.277080141] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
  [component_container_isolated-3] [ERROR] [1758713047.277091694] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
  [component_container_isolated-3] [WARN] [1758713050.019393337] [global_costmap.global_costmap]: KeepoutFilter: Filter mask was not received
  [rviz2-2] [INFO] [1758713053.186857445] [rviz]: Message Filter dropping message: frame 'odom' at time 714,820 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
  [component_container_isolated-3] [WARN] [1758713053.4
  ```

- Notes:
  - The error appears to come from a BT node expecting a `wait_for_service_timeout` value on the blackboard.
  - I am using the default Nav2 behavior tree XML shipped with Humble (not a custom tree).
  - The additional warnings about KeepoutFilter mask and RViz TF timestamps may be separate configuration issues, but including them for completeness.

- Configuration snippets (if relevant):
  - Launch file used: ros2_launch_c nav2_bringup tb3_simulation_launch.py use_simulator:=False map:=/home/usuario/workspaces/signos_ws/map/map.yaml params_file:=/home/usuario/workspaces/signos_ws/src/nav2_bringup/params/nav2_params.yaml  use_sim_time:=True

  - `nav2_params.yaml`: bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    wait_for_service_timeout: 1.0

- Questions:
  - Is `wait_for_service_timeout` expected to be provided in Humble via `bt_navigator` parameters/blackboard for this XML, or is this a regression where the default XML requires a key that isn't set by default?
  - If it should be set by the user, what is the recommended parameter name and scope for Humble? For example, should we add under `bt_navigator.ros__parameters` something like:
    ```
    bt_navigator:
      ros__parameters:
        wait_for_service_timeout: 5.0
    ```
    or is there another canonical way to provide this blackboard entry on Humble?
This was working until yesterday when I installed some behavior tree packages but now just with ros2 sourced it does not work any more

SteveMacenski, 2025-09-26:
I am not able to reproduce in a containerized environment, so I think there's something wrong on your side. The following command works for me fine with the latest Humble binaries. 

> ros2 launch nav2_bringup tb3_simulation_launch.py

Have you pulled in additional BT nodes from other branches and trying to make them work on Humble?

SteveMacenski, 2025-11-19:
Closing due to lack of response