ros-navigation/navigation2 #5551: Add Collision Monitor Toggle Service as Simple Commander Python API

SteveMacenski, 2025-09-25:
https://github.com/ros-navigation/navigation2/pull/5493 and https://github.com/ros-navigation/navigation2/pull/5532 introduce the ability to toggle on/off the collision monitor when you want to make contact with things like docks or other applications. We have behavior tree nodes to do so, but to round it off it would be nice to have a Python3 API for it as well

armgits, 2025-09-26:
I'm interested in taking this issue, I'll see if I can come up with something to begin with next week!

DavidG-Develop, 2025-09-28:
Oh crap, I did not see you already opened an issue for this.... After we discussed in #5532 I went directly into implementing it not checking back the issues... I already opened a PR for this... #5559

SteveMacenski, 2025-09-29:
Let me know if you need any help!

closed 2025-09-29 by SteveMacenski